esphome/feature-requests #1639: Report Dallas "Requested conversion failed"  back

slm0001, 2022-02-17:
**Describe the problem you have/What new integration you would like**

I would like to check, whether the Dallas DS18B20 is connected or not?
In Arduino envorinment I could check that, but in ESPHome I did not find the corresponding function!
Overheating is pretty risky in my project so I have to be sure that the sensor values are valid. 
My request is that in this case also get a "NAN" or any predefined value.

**Please describe your use case for this integration and alternatives you've tried:**

Right now if during runtime I disconnect a sensor, the measured value remains. 

**Additional context**

nagyrobi, 2022-02-17:
Add this filter to your dallas sensor:
```
    filters:
      - filter_out: nan
```

slm0001, 2022-02-17:
filter_out removes every nan value, doesn't it?
However I  would like to have a nan value, when the sensor is removed after boot.
I will give a shot and let you know the result...

slm0001, 2022-02-17:
I have checked the source code: (https://esphome.io/api/dallas__component_8cpp_source.html)
In row 112 I would like to add this line:

sensor->publish_state(NAN);

I think that would solve my issue....

nagyrobi, 2022-02-17:
would be nice to have it as a configurable option.

closed 2022-02-19 by OttoWinter