esphome/feature-requests #1641: Add `optimistic` to modbus select

nagyrobi, 2022-02-17:
**Describe the problem you have/What new integration you would like**
The new [Modbus Controller Select](https://esphome.io/components/select/modbus_controller.html) is great, but it only misses one property, which is available in case of Template Select: `optimistic`.

The problem is that the reported state of the component only updates when the main modbus controller updates the entire set of data, which, depending on the controlled device, can take a long time. 
There is of course `update_interval` which can be set to adjust these timings on the controller itself but that value is limited to what the hardware being controlled is able to report.

Since it's likely that the reported value of the select will be the same as the selected state of it, it would make sense to have an option to report the value immediately, just like `optimistic` is for the template select.

The real update of the main component would overwrite (usually with the same value) when the time will come - and that would be just fine.

**Please describe your use case for this integration and alternatives you've tried:**
I am using automations which trigger when the reported state of the select changes. These always come in too late and interferes with normal user interface operations.

It's not possible to decrease `update_interval` for the main controller less than 6 seconds on this certain device as it locks up due to polling too often. An ideal situation would be to report the select back instantly, and let it be refreshed loosely later with the same real value whenever the time comes.

**Additional context**
@stegm @martgras

stegm, 2022-02-21:
@martgras Any objections adding `optimistic`?

martgras, 2022-02-22:
No sounds like a good idea.

nagyrobi, 2022-03-01:
@stegm any news perhaps?

stegm, 2022-03-02:
> @stegm any news perhaps?

Sorry, not yet. Maybe in the next couple of days.

stegm, 2022-03-06:
@nagyrobi I prepared a PR #3267 but it is not yet tested. Do you have the chance to test it?

nagyrobi, 2022-03-06:
Hi @stegm I try to test with:
```
external_components:
- source: github://pr#3267
  refresh: 0s
  components: ["modbus_controller"]
```
But build fails with tons of errors like:

```
INFO Reading configuration /config/esphome/szellozes-illat.yaml...
INFO Updating https://github.com/esphome/esphome.git@pull/3267/head
ERROR Unable to load component modbus_controller.binary_sensor:
Traceback (most recent call last):
  File "/esphome/esphome/loader.py", line 162, in _lookup_module
    module = importlib.import_module(f"esphome.components.{domain}")
  File "/usr/lib/python3.9/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1030, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1007, in _find_and_load
  File "<frozen importlib._bootstrap>", line 986, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 680, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 790, in exec_module
  File "<frozen importlib._bootstrap>", line 228, in _call_with_frames_removed
  File "/config/esphome/.esphome/external_components/2df6d485/esphome/components/modbus_controller/binary_sensor/__init__.py", line 32, in <module>
    binary_sensor.binary_sensor_schema(ModbusBinarySensor)
AttributeError: module 'esphome.components.binary_sensor' has no attribute 'binary_sensor_schema'
```

Can you please suggest a different way to test maybe?

nagyrobi, 2022-03-06:
OK, my tests above were with release and beta v2022.2.6.
With DEV it builds fine, and optimistic works great!

closed 2022-03-09 by jesserockz