esphome/feature-requests #1643: Prebuild images on container creation for faster "Update all" operation

jsBlackBelt, 2022-02-19:
**Describe the problem you have/What new integration you would like**

Currently, Update all takes a huge amount of time, since images are built at the moment, each taking a few minutes and for systems with many devices this mounts up to a very long time.

**Suggestion:**
On ESPHome upgrade or container create - pre-build all device images in the background, so they are ready to upload to the device. Then when the user clicks "Update all" it is simply a matter of the actual upload to device which takes much less time.

**Please describe your use case for this integration and alternatives you've tried:**

User clicks "Update all" and then ESPHome starts building images one by one, user has to wait a long time to know if the process was successful or not for each device.

**Additional context**

nagyrobi, 2022-02-19:
+additional benefit as an extra side feature extra to this, one could save a "snapshot" of all the binaries by downloading them in one package and save them on the local machine for future backup/revert to previous state if something goes wrong with the new release.

jsBlackBelt, 2022-10-08:
This could be a `good first issue` or part of `WTH` or hacktoberfest?
I could pick a stab at it, with some directions

nagyrobi, 2022-10-08:
Go ahead!

jsBlackBelt, 2022-10-08:
@nagyrobi 
So what I am thinking is:
* Create a new websocket command `EsphomeUpdateAllOfflineHandler` that will create the bin files and store them in the config directory
* Modify `EsphomeUpdateAllHandler` to first check for their existence before creating them and if they exist, just move on to flashing them. 

1. Does that make sense to you?
2. I could not find the actual logic that runs when a handler is invoked.

nagyrobi, 2022-10-08:
- I'd suggest a separate subdirectory for the bins (don't mix with config yamls)
- About checking their existence, I'd suggest to always overwrite them when going through this new route manually.
- Have immediate flashing optional. Say, you'd just want to build all the bins for putting aside without mass flashing.