esphome/feature-requests #1644: ESP32 cosine generator

sunshine-nick, 2022-02-19:
**Describe the problem you have/What new integration you would like**

ESP32 has a built in cosine generator see  https://github.com/krzychb/dac-cosine  I would like to add this to toESPHome and use it like ledC pwm  I have the relevant ESP32 and IDF experience but have not compiled ESPHome from scratch. Is there anyone who has added the cosine generator already or is there anyone that would like to help me do this.
**Please describe your use case for this integration and alternatives you've tried:**

I need to use a sine wave (10 -50KHz) to for generating AC signals, neasuring response and  applying ouput to opamp gain circuits. I would use this in a very similar way to using the  ESP32 ledC output. API. Entity state is on/off and attributes would be chan nr ,  frequency and possibly amplitude
**Additional context**

I have 5 years low level ESP32 experience with the ESP IDF and would like to compile ESPhome in IDF mode but have not got any experience with this and would like some help in setting up the correct tool chain

nagyrobi, 2022-02-19:
Excellent idea!

huVVer, 2022-02-20:
Here is a test program I finished yesterday in the Arduino IDE.  Note that the DAC outputs need to be buffered to work properly.  I used MCP6001UT rail-rail input/output OpAmps for this, and scale the outputs to 5V.
========================================
  #include <driver/dac.h>
  
  dac_cw_config_t dac_cw_config;

  bool leftright = true; // simple way to demonstrate left or right channels
  uint32_t frequency = 131;
  uint8_t Left = 0, Right = 0;
  static const int i2s_num = 0; // i2s port number

  // 
  // Stereo tones.  4 levels (1 - 4) of logarithmic tones, plus tone OFF (0) per channel.
  //
  void Tones (uint32_t frequency, int8_t levelLeft, int8_t levelRight){
    
    dac_cw_scale_t Scale;
    
    switch (levelLeft) {
      case 0: 
      dac_output_disable(DAC_CHANNEL_1);  // off
      break;

      case 1: 
      dac_output_enable(DAC_CHANNEL_1);
      Scale = DAC_CW_SCALE_8; // quietest
      break;

      case 2: 
      dac_output_enable(DAC_CHANNEL_1);
      Scale = DAC_CW_SCALE_4;
      break;
     
      case 3: 
      dac_output_enable(DAC_CHANNEL_1);
      Scale = DAC_CW_SCALE_2;
      break;
       
      case 4: 
      dac_output_enable(DAC_CHANNEL_1);
      Scale = DAC_CW_SCALE_1; // loudest
      break;

    default:
      dac_output_disable(DAC_CHANNEL_1);
      break;
    }

      dac_cw_config = {
        .en_ch  = DAC_CHANNEL_1,
        .scale  = Scale,
        .phase  = DAC_CW_PHASE_0,
        .freq = frequency,
        .offset = 127
      };
      
      dac_cw_generator_config (&dac_cw_config);
    
    switch (levelRight) {
      case 0: 
      dac_output_disable(DAC_CHANNEL_2);  // off
      break;

      case 1: 
      dac_output_enable(DAC_CHANNEL_2);
      Scale = DAC_CW_SCALE_8; // quietest
      break;

      case 2: 
      dac_output_enable(DAC_CHANNEL_2);
      Scale = DAC_CW_SCALE_4;
      break;
     
      case 3: 
      dac_output_enable(DAC_CHANNEL_2);
      Scale = DAC_CW_SCALE_2;
      break;
       
      case 4: 
      dac_output_enable(DAC_CHANNEL_2);
      Scale = DAC_CW_SCALE_1; // loudest
      break;

    default:
      dac_output_disable(DAC_CHANNEL_2);
      break;
    }
    
      dac_cw_config = {
        .en_ch  = DAC_CHANNEL_2,
        .scale  = Scale,
        .phase  = DAC_CW_PHASE_0,
        .freq = frequency,
        .offset = 127
       };
       
       dac_cw_generator_config (&dac_cw_config);
  }; 
  
void setup() {
  dac_output_enable(DAC_CHANNEL_1);
  dac_output_enable(DAC_CHANNEL_2);
  dacWrite(25, 127); // set zero point
  dacWrite(26, 127); // set zero point 
  dac_cw_generator_enable();
  Serial.begin (115200);
  delay (100);
}

void loop() 
    delay (62);

    Tones (frequency, leftright, !leftright);
  
    frequency += 20;
    if (frequency > 1500){
      frequency = 131;
      leftright = !leftright;

      delay (500);
    }
  }

sunshine-nick, 2022-02-21:
Hello HuVVer,
Thanks alot. I shall be testing this later today.  I know there is an example in the ESPhome dpcs on how to add Arduino style extensions so I will be studying that first. Thats the first hurdle for me  as I normally only use the IDF command line, although I have tried a few arduino  style ESP32 projects in the past to see how it works. So I understanmd the code you have written.

Just as a matter of interest, when I add this code through an ESPhome YAML script and compile how does this affect Platform IO dependancies for the DAC interface, Is platform IO just included in the ESPHome toolchain and works more or less transparently ?

Thanks alot for your help.

sunshine-nick, 2022-02-23:
Hi, well I solved my problem.
I went back to the original example and checked some of the incloude files in hal_dac.c and moved the C code into a emprty ESPhome custome component, and he prestot ot worked.
 This is my ESPhome yamal file:

vvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvv
esphome:
  includes:
    - my_custom_sensor.h
    
  name: ttgotest1

esp32:
  board: esp32dev
  framework:
    type: arduino

# Enable logging
logger:

# Enable Home Assistant API
api:
  password: ""

ota:
  password: ""

wifi:
  ssid: "work2"
  password: "tangray21"

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Ttgotest1 Fallback Hotspot"
    password: "nZfEshRRytzU"

captive_portal:

light:
  - platform: status_led
    name: "ttgo1 led"
    pin: GPIO5

# Example configuration entry
sensor:
- platform: custom
  lambda: |-
    auto my_sensor = new MyCustomSensor();
    App.register_component(my_sensor);
    return {my_sensor};

  sensors:
    name: "My Custom Sensor"^

^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^

And this is my C/C++ test case.

vvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvv

#include "esp32-hal-dac.h"
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "rom/ets_sys.h"
#include "esp_attr.h"
#include "esp_intr.h"
#include "soc/rtc_io_reg.h"
#include "soc/rtc_cntl_reg.h"
#include "soc/sens_reg.h"
#include "soc/rtc.h"

[... 128 lines cut ...]
  publish_state(count);
  count  = count +1;
  if (count == 40) count = 10;
    
  // frequency setting is common to both channels
  dac_frequency_set(clk_8m_div, frequency_step);
  /* Tune parameters of channel 2 only
   * to see and compare changes against channel 1
   */
  dac_scale_set(DAC_CHANNEL_2, scale);
  dac_offset_set(DAC_CHANNEL_2, offset);
  dac_invert_set(DAC_CHANNEL_2, invert);
  
  scale++;
  if (scale == 4) scale = 0;

  float frequency = RTC_FAST_CLK_FREQ_APPROX / (1 + clk_8m_div) * (float) frequency_step / 65536;
  
  sprintf(sbuf,"clk_8m_div: %d, frequency step: %d, frequency: %.0f Hz\n", clk_8m_div, frequency_step, frequency);
  ESP_LOGD("TEST: ",sbuf);
  sprintf(sbuf,"DAC2 scale: %d, offset %d, invert: %d\n", scale, offset, invert);
  ESP_LOGD("TEST: ",sbuf);  
  }
  
};

sunshine-nick, 2022-02-23:
So I get this feature request issue is closed.

closed 2022-02-23 by sunshine-nick