esphome/feature-requests #1646: M-Bus support

poear, 2022-02-20:
**Describe the problem you have/What new integration you would like**
I have a energy meter installed in my house that exposes its data via M-Bus. I would like to see support for reading values over M-Bus. Since M-Bus, supposedly, is used for many kinds of utility meters this might be useful to a lot of people in Europe.
I'm looking for some guidance on how to continue. I'm a bit lost or stuck. If anyone has some input or

**Please describe your use case for this integration and alternatives you've tried:**
I want to enable users to easily add new components for M-Bus meters.

Using M-Bus Master hardware ([MIKROE-3880](https://www.mikroe.com/m-bus-master-click) looks promising, or [DIY version](https://github.com/rscada/libmbus/tree/master/hardware)) and code supporting communication with M-Bus like [libmbus](https://github.com/rscada/libmbus). Adding support for a new meter should be limited to implementing the [manufacturers protocol](https://gavazzi.se/app/uploads/2020/11/em111_em112_m1_cp.pdf).

Alternatives:
I have not tried anything else.

**Additional context**
[Wikipedia says](https://en.wikipedia.org/wiki/Meter-Bus)
_M-Bus or Meter-Bus is a European standard (EN 13757-2 physical and link layer, EN 13757-3 application layer) for the remote reading of water, gas or electricity meters. M-Bus is also usable for other types of consumption meters, such as heating systems or water meters. The M-Bus interface is made for communication on two wires, making it cost-effective. A radio variant of M-Bus Wireless M-Bus is also specified in EN 13757-4._

_The M-Bus was developed to fill the need for a system for the networking and remote reading of utility meters, for example to measure the consumption of gas or water in the home. This bus fulfills the special requirements of remotely powered or battery-driven systems, including consumer utility meters. When interrogated, the meters deliver the data they have collected to a common master, such as a hand-held computer, connected at periodic intervals to read all utility meters of a building. An alternative method of collecting data centrally is to transmit meter readings via a modem._

_Other applications for the M-Bus such as alarm systems, flexible illumination installations, heating control, etc. are suitable._

github-k8n, 2022-07-21:
Dumping some information here as this was a bit difficult to find and maybe might help someone looking for this.

At the moment I am using a raspberry pi to read out the data from a UART IR head (same the ones used for the electricity meters used with SML) connected to the heat meter (SENSUS Pollustat) using pyMeterBus https://github.com/ganehag/pyMeterBus/ and sending the data on via MQTT (using paho-mqtt) taking inspiration from the scripts here: https://forum.iobroker.net/topic/32722/smart-meter-sensus-pollucom-e/4  (note the initial wakeup sequence for these kind of meters) , however it would be great if there were a way to incorporate this somehow in esphome. (read-only as pyMeterBus does might be a good start?)

Maybe if someone could refactor the code from pyMeterBus into C to make it usable as a module in esphome?

(of course, full support of M-Bus with all the bells and whistles might be even better :)

Aikawa24, 2022-10-08:
some guy on
[https://www.mikrocontroller.net/topic/113984](url)

make this code to read the pollucom E via opto ir

`#include <stdio.h>
#include <fcntl.h>
#include <sys/stat.h>

char *buf;
char in[256] = "";
ssize_t ret_out;
int fd = open ("/dev/ttyUSB0", O_RDWR | O_NOCTTY );
set_interface_attribs (fd, B2400, 0);  // set speed to 2,400 bps, 8n1 (no parity)
set_blocking (fd, 0 );                // set no blocking
strcpy( in, "UUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUU");
ret_out = write (fd, in, strlen(in));           // sent wakeup
usleep ((1000 + 25) * 100);             // sleep enough to transmit plus
buf = read_data( fd);
// Initialisiere MBus -> ACK mit 'E5'
in[0] = 0x10;
in[1] = 0x40; // Init MBus
in[2] = 0x00;
in[3] = 0x40; // Checksumme = in[1] + in [2]
in[4] = 0x16;
ret_out = write (fd, in, 5);           // sent
usleep ((1000 + 25) * 100);             // sleep enough to transmit plus
buf = read_data( fd);
if (buf[5] == 0xE5)
{ 
// Get Data from MBus
in[0] = 0x10;
in[1] = 0x5B;  // GetData
in[2] = 0x00;
in[3] = 0x5B;   // Checksumme = in[1] + in [2]
in[4] = 0x16;
ret_out = write (fd, in, 5);
usleep ((1000 + 25) * 100);             // sleep enough to transmit plus
buf = read_data( fd);
// Zaehlerstand
double waerme = ((double) toInt(&buf[26])) + (100.0 * (double) toInt(&buf[27])) + (1000.0 * (double) toInt(&buf[28])) + (10000.0 * (double) toInt(&buf[29]));
//aktuelle Leistung
double leistung = ((double) toInt(&buf[44])) + (100.0 * (double) toInt(&buf[45])) + (1000.0 * (double) toInt(&buf[46])) + (10000.0 * (double) toInt(&buf[47]));
}

int toInt( char *in ) {

char out[2];
int iOut = 0;
sprintf(out, "%02x", in[0]);
iOut = atoi(out);
return iOut;
}`

karllinder, 2022-11-21:
Hi

Looks like there is some need for this, this is what @turbokongen has solved in HA
https://github.com/turbokongen/hass-AMS

There is also a mbus.h that I use, not working 100%, taken from this post:
https://www.hjemmeautomasjon.no/forums/topic/5141-han-til-home-assistant-over-esphome/

wmeler, 2023-03-22:
One more interesting link https://github.com/dev-lab/esp-iot-mbus

bf8392, 2023-04-29:
> yep, would be great to have (wired)Mbus support!

Wm-bus means wireless m-bus :-) but it would be best to have support for both... someone made a "hacky" integration..maybe it helps for the standard integration... https://community.home-assistant.io/t/hacky-integration-for-m-bus/127288/14

RogMoe, 2023-05-09:
MBus-master adapters<>USB are very cheap and failsafe. I prefer hardwired instead of wireless.

mortenx, 2024-04-16:
+1 for MBus Support

kluszczyn, 2024-06-16:
There is custom component based on CC1101 transceiver implementing WMBus reader
https://github.com/SzczepanLeon/esphome-components

luka6000, 2024-07-10:
> There is custom component based on CC1101 transceiver implementing WMBus reader https://github.com/SzczepanLeon/esphome-components

I'm also using this with great success for electricity, heat and water meters (total 4 independent meters with multiple data points). It can be used standalone or with wmbusmeters add-on in HA (passing raw datagrams - how awesome is that!)
Buy frankly speaking I don't like this component implementation. It works great but it's purpose built all-in-one. 
@jesserockz what would you suggest? Wireless M-bus implementation requires receiver component (like CC1101), then protocol implementation (M-bus) and then meters definition with specific attributes/sensors. Would you accept such a all-in-one approach or should it be rebuilt with layers?

JeeCee1, 2025-01-21:
It's not ESPHome, but at least on ESP: https://github.com/Zeppelin500/MBusino