esphome/feature-requests #1653: Add timeout for deep_sleep.prevent

mkocus, 2022-02-23:
**Describe the problem you have/What new integration you would like**
It would be useful to have an timeout option for deep_sleep.prevent action, so it would prevent deep sleep only for a given time.

**Please describe your use case for this integration and alternatives you've tried:**
The use case is quite simple: I want the touchscreen (and whole esp) to stay awake for x seconds after last touch, so every touch event invokes the deep_sleep.prevent. But in the current deep_sleep implementation it prevents sleeping until deep_sleep.enter is called.

**Additional context**
I've actually already implemented all the code required in local deep_sleep component (both python config, validation and cpp/h), tested it locally and it works quite well.

But wanted to consult the community / component maintainer if it is really needed and/or maybe there is some other solution in the current implemented I somehow missed.

nagyrobi, 2022-07-19:
> I've actually already implemented all the code required in local deep_sleep component

Please make a PR so the devs can analyse your solution.

cdwertmann, 2022-08-17:
Another way to implement this could be to create an Action that allows reconfiguring deep_sleep, e.g.
```
on_...:
  then:
    - deep_sleep.set:
        id: deep_sleep_1
        sleep_duration: 20min
```

afarago, 2024-06-01:
It would also be an alternative solution to "reset" a scheduled deep_sleep.

I would envision such an action: `deep_sleep: reset` - this could call the same code as it is in the :setup call.
```cpp
if (run_duration.has_value()) {
  ESP_LOGI(TAG, "Scheduling Deep Sleep to start in %" PRIu32 " ms", *run_duration);
  this->set_timeout(*run_duration, [this]() { this->begin_sleep(); });
}
```
See also: https://www.reddit.com/r/Esphome/comments/1cen755/how_to_reset_gpio_deep_sleep_countdown/