esphome/feature-requests #1654: I need some help controlling my BLE elevation-bed :) ???

jascdk, 2022-02-24:
**Describe the problem you have/What new integration you would like**

I have an elevation-bed in our bedroom. It is equipped with Danish Branded Linak actuators/motors, which can be controlled through an IOS/Android APP. It communicates through BLE. They are fitted with LED lights underneath for night lightning.

I want to know, if it is possible to use an ESP32 to control it - maybe the "cover" component could be useful - I don't know yet - so far. 

The thing I have tried so far is turning the lights on and of. 

I connected the app to my Mac and installed a packet-logger (part of Xcode). With that I could sniff the commands for turning the light on and of. 

To do so - I downloaded the nrf Connect App on my iPhone. Connected to the bed - went to client mode in the app and wrote an request (Byte-Array) to the first "unknown characteristic". 

When I wrote the value 9200 the light went on - and with 9300 it went of - so far so good:) 

Would it be possible to use ESPhome to send a command like this via a switch, so I can automate it?

**Please describe your use case for this integration and alternatives you've tried:**

**Additional context**

I have attached some pictures of the nrf app.

[image]
[image]

emileberhard, 2022-07-14:
Hey! I'm trying to do something very similar. Did you ever make any progress?

nagyrobi, 2022-07-14:
Please contact our Discord channel for this kind of support discussions. There's plenty of helpful people there.

closed 2022-07-14 by nagyrobi

emileberhard, 2022-07-14:
Oh wait sorry I found it

jascdk, 2022-07-31:
> Hey! I'm trying to do something very similar. Did you ever make any progress?

Hi nope - never came any way further - how about you? :)