esphome/feature-requests #1656: A method to reliably check a live connection with HA

benjaminvdb, 2022-02-25:
**Describe the problem you have/What new integration you would like**

I'd like a method that I can reliably use to ensure (i.e. make it more likely) that a service call from ESPHome will reach HA. The current [`api.connected` condition](https://esphome.io/components/api.html#api-connected-condition) doesn't seem to be reliable enough for my use-case because of various reasons:

1. It makes no distinction between HA and other connections such as remote logs or (I assume) any other client e.g. aioesphomeapi.
2. `api.connected` makes use of a list of clients which, I assume, may contain stale connections. This means that if HA had successfully connected and then disappears from the network, this isn't picked up by ESPHome. In that case `api.connected` evaluates to `True` even though there are no active connections and service calls to HA won't be picked up.

It would be great if a means of checking if the device has a live connection to HA. I can then utilize this to use fallback logic in case there's no such connection. I'm not the best person to come up with a suggestion, since I'm not familiar with the codebase, but I imagine something along the lines of:

- Periodically clean out the list of clients, e.g. by requiring them to send a 'keep alive' signal every now and then and removing clients if such signal hasn't been received for a certain period of time.
- Implement client entity types that can be set when connecting to the ESPHome API.
- These can then be used to differentiate between HA or other types of clients. A new `home_assistant.is_connected` condition could then be implemented by checking if a HA client is present in the list of clients.

**Please describe your use case for this integration and alternatives you've tried:**

I have Shelly 1L relay connected to a toggle/rocker switch. The Shelly is set in "detached mode", i.e. the relay is always on and the state change of the switch is used to send a `light.toggle` service call to HA to toggle a Zigbee smart bulb, which is connected to the output of the Shelly. However, if HA becomes unreachable for some reason, it would be great if the smart bulb could be toggled using the relay. Below you can find my config, which is basically an adapted version of a [Shelly 1 config in the documentation](https://www.esphome-devices.com/devices/Shelly-1).

<details><summary>Shelly config</summary>

```
substitutions:
  device_name: kitchen

# Basic Config
esphome:
  name: ${device_name}
  platform: ESP8266
  board: esp01_1m

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  ap:
    ssid: ${device_name}
    password: !secret wifi_password

logger:
api:
ota:

web_server:
  port: 80

#==============================================================
switch:
  - platform: gpio
    pin: GPIO5
    id: shelly_1l_relay

binary_sensor:
  - platform: gpio
    name: ${device_name} Input
    pin:
      number: GPIO4
    # small delay to prevent debouncing
    filters:
      - delayed_on_off: 50ms
    # config for state change of input button
    on_state:
        then:
          - if:
              condition:
                and:
                  - wifi.connected:
                  - api.connected:
                  - switch.is_on: shelly_1l_relay
              # toggle smart light if wifi and api are connected and relay is on
              then:
                # - switch.toggle: shelly_1l_relay
                - homeassistant.service:
                    service: light.toggle
                    data:
                      entity_id: light.kitchen_light
              # else, toggle relay
              else:
                - switch.toggle: shelly_1l_relay
    id: button
```

</details>

The way I test this setup is that I turn on the smart light and then disconnect HA from the network by pulling the ethernet plug. I was expecting `api.connected` to evaluate to `False`, but it doesn't. I think this is because `api.connected` wasn't intended to be used this way, i.e. it's a feature, not a bug.

**Additional context**

_I've tried my best to investigate this by reading the documentation and checking the ESPHome source code. However, I'm not at all familiar with the codebase. My apologies if I made any wrong assumptions about how the code works. I've tried to emphasize where I made these assumptions and I hope they'll help the reader understand what I'm getting at._

benjaminvdb, 2022-03-03:
I've found a way to implement a decently reliable fallback mode by pinging HA using an external component. If all packets got dropped, the fallback mode kicks in. I've left my config on the [community forums](https://community.home-assistant.io/t/fall-back-to-relay-if-not-connected-to-ha/396459/11).

This method isn't optimal, as it relies on an external component and there's a delay between the disconnect and when it gets detected. A cleaner way of doing this would be highly appreciated.

ssieb, 2022-08-25:
There aren't stale connections.  There is a ping on the connection, so if the other end isn't responding, it will get disconnected.

benjaminvdb, 2022-08-26:
> There aren't stale connections. There is a ping on the connection, so if the other end isn't responding, it will get disconnected.

Thanks for clarifying this!

Point (1) still stands, i.e., ESPHome makes no distinction between HA and loggers (as also stated in the [documentation](https://esphome.io/components/api.html#api-connected-condition)), making it impossible to utilize `api.connected` to trigger alternative behavior when the connection with HA is lost.

nagyrobi, 2023-02-09:
See for a workaround: https://github.com/esphome/feature-requests/issues/2108#issuecomment-1423905640

Ulrar, 2023-02-09:
Also I don't know what the timeout is set to, if it exists, but it's not very fast. I've had a lot of stability issues with HA 2023.2 so these last few days it's been very apparent to me that if `api.connected` goes to false at some point, it's really not quick enough to be usable in the above use case (light switch fallback)

nagyrobi, 2023-02-09:
See the workaround above, you can set short timeout with fast keepalive toggles/and shorter script delays.

0x3333, 2023-02-09:
> @0x3333 @poldim See for a workaround: [#2108 (comment)](https://github.com/esphome/feature-requests/issues/2108#issuecomment-1423905640)

I got a hard time trying to understand why I got mentioned 🤣🤣🤣

Here's why: https://github.com/esphome/feature-requests/issues/1066

@nagyrobi, thanks!

nagyrobi, 2023-03-05:
Some backround info on how ping-pong between HA and ESPHome nodes work:
https://github.com/esphome/aioesphomeapi/pull/395#issuecomment-1454831563
https://github.com/home-assistant/core/issues/89047#issuecomment-1454911365

LordMike, 2023-03-11:
I'm making a relay that will turn a heating pad on/off to keep some batteries warm during the colder months, and I'm very concerned about _something_ messing this up meaning the heating pad will remain on. So I'm brainstorming issues and solutions for it. My current concerns are that if I f.ex. forget all about this relay, and change an IP address or clear out devices, my esphome device won't be able to pull the "current temperature" from HA and make its decision on turning the heat on/off, potentially leaving it on.

I asked ChatGPT, because I had a hard time with the docs, and it suggested a (non existent) automation named `on_failure`, to be used like this:

```
sensor:
  - platform: homeassistant
    id: my_sensor
    entity_id: sensor.my_homeassistant_sensor
    on_failure:
      then:
        - lambda: |-
            id(my_sensor).publish_state(25.0);
```

I was happy, because it seems like it could solve my problem: If a sensor values goes stale or fails updating, I can force it locally to some value (here 25), which in turn can do something with my logic on esphome, such as saying "all good, turn off".

`on_failure` doesn't exist, sadly, but it begs the question. HA has the concept of "expire_after" which means a sensor value is unavailable if it isn't updated every N seconds. Could something similar be applied in esphome, maybe a `on_stale` or `on_unavailable` ?

nagyrobi, 2023-03-11:
For such mission critical tasks don't rely on HA forwarding the temp for you. Use a locally connected Dallas sensor. I use one on a 50 meters long CAT5e cable, it's working since 3 years.

ssieb, 2023-03-16:
You could use a timer script triggered by the `on_value:` of the sensor.

nagyrobi, 2023-10-30:
Implemented in https://github.com/esphome/esphome/pull/5628

With `on_client_connected` and `on_client_disconnected` triggers one could check `client_info` variable for `Home Assistant` string or `client_address` for the address of HA, and set thus determine if the client which has connected, is HA or something else.

closed 2023-10-30 by nagyrobi

Ulrar, 2023-10-30:
Amazing news, thank you very much!

nagyrobi, 2024-01-15:
For reference:

```
api:
  ...
  on_client_connected:
    - logger.log:
        format: "Client %s connected to API with IP %s"
        args: ["client_info.c_str()", "client_address.c_str()"]
    - if:
        condition:
          lambda: 'return (0 == client_address.compare(std::string{"your.ha.fixed.ip"}));'
        then:
          - ... #do something
```