esphome/feature-requests #1658: Midea component for ethernet ESP32 board like Wireless Tag WT32-ETH01 and similar

jerzyquercum, 2022-02-26:
**Describe the problem you have/What new integration you would like:**

I want to compile ESPHome configuration on Wireless Tag WT32-ETH01 board with Ethernet connection enabled and I cannot because "Component climate.midea requires component wifi."  Is it possible to allow the use of midea components without Wi-Fi only on an Ethernet connection?

**Please describe your use case for this integration and alternatives you've tried:**

Midea components work fine on these boards with Wi-Fi enabled (Wi-Fi component enabled).

**Additional context**

Please allow midea components to be installed without Wi-Fi components.

Yours sincerely

Jeremiasz

nagyrobi, 2022-02-26:
WiFi is set as dependency:
https://github.com/esphome/esphome/blob/d99c5ed8901279def13815d9f2b22f9f8a203a9f/esphome/components/midea/climate.py#L38

You could perhaps try to copy the source of this component in your local environment, remove the wifi dependency and try to build it from there. If it works, you could submit a PR with the fix.

jerzyquercum, 2022-03-20:
I installed the Python library and downloaded ESPHome locally.
I removed the wifi dependency in the Climate.py file and built the firmware locally for the Wireless Tag WT32-ETH01, after flashing (via the programmer) and connected to Midea AC everything works ok, all functions are available and works exactly like in Wi-Fi version.
I'll make this suggestion in PR - remove Wi-Fi dependencies or change it to Wi-Fi or Ethernet alternatives.

Yours sincerely

Jeremiah

nagyrobi, 2022-03-21:
There's an issue with that PR. Too many unrelated changes...