esphome/feature-requests #1660: Add support for SSD1309 Display

gohlas, 2022-02-28:
**Describe the problem you have/What new integration you would like**
I bought this Lilygo-TTV and I am trying to use with ESP home but it seem there is no support for this transparent SSD1309 display
https://www.aliexpress.com/item/1005001681414979.html?spm=a2g0o.order_list.0.0.21ef1802XnEQFz

**Please describe your use case for this integration and alternatives you've tried:**
I do manage to download the software, tried using the SSD1306 display over SPI integration but didn't work (I saw in some website that the library would be the same for 1306 and 1309 but no success here).
I don't have the knowledge to adapt it ESPHome, it would be nice to have some nice person helping on this.

**Additional context**
"Oficial" library can be found here:
https://github.com/Xinyuan-LilyGO/LilyGo-TTV/tree/master/lib/U8g2_Arduino

Usage examples for display:
https://github.com/Xinyuan-LilyGO/LilyGo-TTV/tree/master/examples/DISPALY

I found another library that might work:
https://github.com/sparkfun/HyperDisplay_SSD1309_ArduinoLibrary

jerobins, 2022-04-29:
It works with the SSD1306, need to set GPIO33 high to turn the screen on.
`
output:
  - platform: gpio
    pin: GPIO33
    id: 'generic_out'

switch:
  - platform: output
    name: "Screen Enable"
    output: 'generic_out'
`

gohlas, 2022-05-02:
Just tested and actually it works.
Thanks for the help @jerobins!

I will close this then.

closed 2022-05-02 by gohlas