esphome/feature-requests #1661: Add support for fan preset modes

finity69x2, 2022-03-04:
**Describe the problem you have/What new integration you would like**

add a configuration option to allow the use of defined preset modes for fans that support them.

**Please describe your use case for this integration and alternatives you've tried:**

adding preset modes to ESP based fans would allow more flexibility than just supporting percentage speeds.

**Additional context**

nagyrobi, 2022-03-06:
I think this is deprecated from HA side... originally they supported speeds as presets, but it was changed to percentage.

scotty1395, 2022-03-31:
From what I understand, `speed_list:` is deprecated but `preset_modes:` isn't?
ESPHome populates `speed_list: off, low, medium, high` but leaves `preset_modes: null`.

voed, 2023-02-12:
Any progress on this?

millerm1993, 2023-04-06:
I'm also interested in this!

ChristianNewton, 2023-04-08:
> I'm also interested in this!

Me aswell, I would love if this could get some more attention.

Robbe-B, 2023-08-01:
Same, is there already any alternative in the mean time ?

mill1000, 2023-10-31:
> I think this is deprecated from HA side... originally they supported speeds as presets, but it was changed to percentage.

Here's what the HA dev docs say about fan entities and preset modes.
https://developers.home-assistant.io/docs/core/entity/fan#preset-modes

I may take a swing at this if I find enough time.

mill1000, 2023-11-08:
I've taken some steps to start implementing this.

- How does everyone plan to use this function?
- What fan components are you using? (e.g. speed fan, h-bridge, etc)

nagyrobi, 2023-11-08:
I use speed and binary fan.

ChristianNewton, 2023-11-08:
> I've taken some steps to start implementing this.
> 
> - How does everyone plan to use this function?
> - What fan components are you using? (e.g. speed fan, h-bridge, etc)

Hey, thanks for helping work on this! I'm using the Tuya fan component. It works very well, I just wish that there would be an easier way to control it other than 0-100, since it is only 4 speeds.

mill1000, 2023-11-08:
> Hey, thanks for helping work on this! I'm using the Tuya fan component. It works very well, I just wish that there would be an easier way to control it other than 0-100, since it is only 4 speeds.

Does setting [speed_count](https://esphome.io/components/fan/tuya#configuration-variables) not achieve that already?

flyize, 2023-11-17:
> I've taken some steps to start implementing this.
> 
> * How does everyone plan to use this function?
> * What fan components are you using? (e.g. speed fan, h-bridge, etc)

Would this allow me to go back to using low, medium, and high - even though I have a 4 speed fan? That's really what I'm hoping for. Using percentages *only* is unnecessarily confusing to anyone else in my house. Everyone understands low, medium, and high.

mill1000, 2023-11-17:
Yes, although that is considered to be an [improper use of the function](https://github.com/esphome/esphome/pull/5694#issuecomment-1807224011) and is better handled [via other means](https://github.com/esphome/esphome/pull/5694#issuecomment-1811364019) (in Home Assistant).

flyize, 2023-11-17:
While I totally get that I can work my way around this developer-imposed HA limitation, it seems insane. People intuitively understand low, medium, and high (especially for voice assistants!). For more advanced use-cases, use the percentages.

So in short, thank you for possibly bringing this back!

filmkorn, 2023-11-26:
I use a speed fan component in an esphome replacement of the broken brains of a smart air purifier - see https://github.com/filmkorn/air_purifier

> How does everyone plan to use this function?

I'd like to reimplement the preset modes provided by the original integretion (VeSync):
- auto: based on the current air pollution)
- sleep: basically lowest speed setting)
- manual: Which the VeSync integration no longer explicitly supports, but it switches to manual mode when you select one of the three speeds of the air purifier. Unsure where this automation is implemented but I'm thinking this could be done through esphome automations.

[image]

All I want is to have this component to support preset modes as enum on the same entity so I don't have to navigate through 'related' entities to find the mode I have currently implemented as separate entity (select component in esphome).

> What fan components are you using? (e.g. speed fan, h-bridge, etc)

`speed` fan like this

```yaml
output:
  - platform: esp8266_pwm
    pin: D1
    frequency: 1000 Hz
    id: pwm_output

fan:
  - platform: speed
    output: pwm_output
    name: "Air Purifier"
    id: air_purifier   
    speed_count: 10
```

mill1000, 2024-01-03:
FYI this feature was released in ESPHome 2023.12.5, and HomeAssistant just added support in 2024.1.0

flyize, 2024-01-05:
@mill1000 I'm a little confused. Where do I define these presets? I see in the ESPHome docs where they are referenced, but not where they are defined.

mill1000, 2024-01-05:
For a speed_fan or hbridge_fan it would looks something like this
```yaml
fan:
  - platform: speed
    name: "Test Fan"
    id: test_fan
    # ...snip...
    # Define desired preset modes here
    preset_modes: 
      - Auto
      - Sleep
    # Write automations for preset modes here
    on_preset_set:
      then:
        lambda: |-
          if (id(test_fan).preset_mode == "Auto") {
            // Auto preset enabled
          }
          else if (id(test_fan).preset_mode == "Sleep") {
            // Sleep preset enabled
            // Maybe set the speed to super low here
          }
          else {
            // No preset enabled
          }
```

Or if you wanted an auto mode that sets fan speed based on a sensor...
```yaml
sensor:
  - platform: adc
    name: Test Sensor
    # ...snip...
    on_value:
      then:
        lambda: |-
          if (id(test_fan).preset_mode == "Auto") {
            // Set fan speed based on sensor value
            auto call = id(test_fan).make_call();
            call.set_speed(x/10);
            call.perform();
          }
```

nagyrobi, 2024-01-05:
> Or if you wanted an auto mode that sets fan speed based on a sensor...

Unfortunately this doesn't work because as soon as you change the speed from the sensor it automatically exits the `Auto` preset.

Actually this preset thing is pretty useless this way...

mill1000, 2024-01-05:
> > Or if you wanted an auto mode that sets fan speed based on a sensor...
> 
> This is exactly what I need!
> 
> But unfortunately it doesn't work because as soon as you change the speed from the sensor it automatically exits the `Auto` preset. 

You're right, it won't as is. Probably need to revisit.

> Actually this preset thing is pretty useless this way... It exits any preset if you change the speed.
That is actually desired behavior, **if** the change comes from a front end (i.e. Home Assistant changes the speed).

> Additionally it would be nice to have a way to select a default preset (to use when node boots). Currently it starts wit no preset selected, but if you choose a preset you can't go back to "no preset"...

You could add a "None" preset to the list. (Or maybe even an empty string ""?)

filmkorn, 2024-01-05:
> But unfortunately it doesn't work because as soon as you change the speed from the sensor it automatically exits the Auto preset.

I can confirm, the `preset_mode` value is lost when I set the fan speed through `fan.turn_on`.

I'm wondering if the mode could also be shown on the web UI (more of a nice to have)?

mill1000, 2024-01-17:
It's a hack but you might be able to work around the issue by setting preset mode in the call.
e.g.
```yaml
sensor:
  - platform: adc
    name: Test Sensor
    # ...snip...
    on_value:
      then:
        lambda: |-
          if (id(test_fan).preset_mode == "Auto") {
            // Set fan speed based on sensor value
            auto call = id(test_fan).make_call();
            call.set_preset_mode(id(test_fan).preset_mode); // Copy preset mode into call so it won't get cleared
            call.set_speed(x/10);
            call.perform();
          }
```

nagyrobi, 2024-04-30:
The whole concept is wrong.

nagyrobi, 2024-04-30:
NB. not ESPHome devs afaik. 
I have no problem with HA limitations if they want to go a certain route, but since this is technically a separate project, which is also meant to be compatible with other tools (moreover, could be used standalone), this limitation shouldn't apply...

mill1000, 2024-04-30:
Current implementation aside, what is the use case for allowing the user to change speeds and keep a preset active?

nagyrobi, 2024-04-30:
A quick example is above, you have a preset "Auto" where speed itself is set based on some other, automated logic, and you have a preset "Manual" where the speed is set by hand.

mill1000, 2024-04-30:
> A quick example is above, you have a preset "Auto" where speed itself is set based on some other, automated logic, 

Support for this behavior was intended. I don't think jesserockz's comment was meant to invalidate this function.

> and you have a preset "Manual" where the speed is set by hand.

Well that is just not having a preset set at all.

nagyrobi, 2024-04-30:
No, the problem is that there can't be a manual preset. Can't switch to no preset, once a preset is set.
It's all over-complicated and over-constrainted.

mill1000, 2024-04-30:
A manual preset makes no sense. It's not a preset. Why not just set a speed (which will clear the preset)?

I'm not saying it's perfect. I'm just trying to understand the limitations.

nagyrobi, 2024-04-30:
Presets are used to choose between various operation modes, which may or may not involve speed setting. 

There should be an option to remove any kind of limiting relation with the speeds, that's all. Currently this kind of implementation just narrows the possible use cases and inforces only a certain point of view which is simply incorrect for the way people like us would like to use it.
The constraining principle is wrong - that's all.

I'm not saying to go against the HA devs, just allow different way of thining too.