esphome/feature-requests #1663: Add support for uint8 datatypes

remyderuysscher, 2022-03-05:
**Describe the problem you have/What new integration you would like**
Support for UINT8 datatypes, since I have devices which have this datatype. I'm unable to get results with the RAW datatype

**Please describe your use case for this integration and alternatives you've tried:**
`   value_type: RAW
    accuracy_decimals: 0
    lambda: |-
      // Apply your conversation and return a float
      // data is an uint8_t array containing the raw modbus bytes
      return static_cast<float_t>( ((uint8_t)data[0] << 8 ));
     `

**Additional context**

m1ch, 2022-03-07:
Hi @remyderuysscher,

`uint8_t` is a pre-defined datatype and usable in lambdas and in globals.

```
globals:
  - id: test1
    type: uint8_t
    initial_value: '0'

on_press:
  then:
    - lambda: |-
        uint8_t test2 = 0;
```

uint8_t is just a typedef regular used in C(++) and Arduino. With the following code snipped you can define additional data-type in your lambdas: (search for cpp typedef)
```
lambda: |-
  typedef unsigned char my_uint8_t;
  my_uint8_t test3 = 99;
```
If this answers your request, I suggest that you close the issue, if not, please update the request.

remyderuysscher, 2022-03-07:
Thanks for the suggestion @m1ch I'm not sure how read out the value of a modbus holding register in the lamba  function.

Do you have any suggestions? [https://esphome.io/components/sensor/modbus_controller.html#modbus-custom-data](url)

`  - platform: modbus_controller
    modbus_controller_id: wtw
    name: "Relatieve luchtvochtigheid afvoerlucht"
    address: 41017
    unit_of_measurement: "%"
    register_type: holding
    value_type: RAW
    accuracy_decimals: 2
    lambda: |-
      ????

m1ch, 2022-03-08:
Hi,

I tried to figure out what your problem is, but with the information you give, that is impossible.

If you want further help, please write down your entire issue!
I don't think that the humidity, in %, is stored in an uint8 type value. How are the required data represented in the sensor? 

Also, this is not a help forum. Don't write a feature request, that implies a change of ESP-home, before you have understood your issue. There is a very active community on discord, or ask your question on Reddit.

remyderuysscher, 2022-03-08:
> Hi,
>
> I tried to figure out what your problem is, but with the information you give, that is impossible.
>
> If you want further help, please write down your entire issue! I don't think that the [...]

I understand and I've posted this issue on the HA forum and tweakers but no solution. 

I have everything working for as UINT16 and FLOAT datatypes. Which work with ESPHome natively. 
Only INT8 is missing, so if can convert INT8 with lamba function to an U_WORD datatype this would suffice I can change this isuue regarding to modbus since uint8_t is supported with lamba. 

BTW Humidity is and UINT8 datatype, this is written in the Honeywel documentation. I would expect FLOAT.

m1ch, 2022-03-08:
> I understand and I've posted this issue on the HA forum and tweakers but no solution.
Link?

> BTW Humidity is and UINT8 datatype, this is written in the Honeywel documentation. And I would expect FLOAT too.
What device do you want to use? This seems to me to be crucial information if you want to solve this. 

The `value_type` usually determines the interpretation of the received data. According to the documentation, it is a convention for modbus to always use a multiple of 16bit words, thus there is no 8-bit type available. However, it is no problem to read the data as a U_WORD and filter them. 

I assume, that the data are 8-bit and that they are at address 0xA039 (41017). I assume further that the values 0-255 correspond to 0-100%. 

```
    platform: modbus_controller
    modbus_controller_id: wtw
    name: "Relatieve luchtvochtigheid afvoerlucht"
    unit_of_measurement: "%"
    value_type: U_WORD
    accuracy_decimals: 2
    lambda: |-
      uitn8_t h = data[0]; // or data[1], depending on, if the device use big, or little-endian
      float humidity = h/255 * 100;
      return humidity;
```

remyderuysscher, 2022-03-08:
> The `value_type` usually determines the interpretation of the received data. According to the documentation, it is a convention for modbus to always use a multiple of 16bit words, thus there is no [...]

Thanks so much, this works. Typo in the uint8_t h declaration ;-) 

Hopefully this thread is useful for others struggeling with 8bit datatypes!

m1ch, 2022-03-09:
@remyderuysscher you are welcome