esphome/feature-requests #1665: Option to use an AddressableLight in a lambda

m1ch, 2022-03-06:
**Describe the problem you have/What new integration you would like**

For an addressable light, in example neopixlebus, I can define an addressable-lambda-effect. In this lambda, I do have the AddressableLight object as "it", thus it is possible to create animation, etc. 
- https://esphome.io/components/light/index.html#addressable-lambda-effect

The same light in a script returns via `id(light_id)` an AddressableLightState object and thus only the basic light operations (turn_on/ off, toggle, ..) are accessible in the script. 

**Please describe your use case for this integration and alternatives you've tried:**

I use an e-ink screen together with a rotary encoder. The e-ink screen is not responsible enough to give me good feedback for my inputs, thus I want to use an additional RGB ring, that updates with every `on_value:` call. 
For this to work, I need to have the AddressableLight object in the lambda. 

Below I added an example how I imagine it could look like.

**Additional context**

Example yaml-configuration:

```
light:
  - platform: neopixelbus
    id: rgb_led

script:
  - id: animate
    then:
      - lambda: |-
          static AddressableLight my_light = id(rgb_led).light();
          static boolean initial_run = true;
          static uint16_t progress = 0;
          uint16_t changes = 0;

          if (initial_run) {
              progress = 0;
              my_light.all() = Color::BLACK;
              initial_run = false;
              return;
          }
          for (int i = my_light.size() - 1; i > 0; i--) {
              my_light[i] = my_light[i - 1].get();
          }
          my_light[0] = Color::random_color();
```

m1ch, 2022-03-07:
The following example shows how I got it to work with a `rotary_encoder` and `addressable_lambda`. If I could use the AddressableLight-object in the on_value-lambda of the rotary_encoder, it would not be necessary to expose the "Rotate" effect to HomeAssistant and the code could be leaner. 

```
globals:
  - id: effect_rotate_running
    type: boolean
    initial_value: "false"

sensor:
  - platform: rotary_encoder
    id: rot_enc1
    internal: True
    name: "${devicename_upper} Rotary Encoder"
    pin_a: 
      number: 18
      inverted: False
      mode: INPUT_PULLUP
    pin_b: 
      number: 17
      inverted: False
      mode: INPUT_PULLUP
    on_value:
      then:
        - lambda: |-
            if ( !id(effect_rotate_running)){
              auto call = id(rgb_led).turn_on();
              call.set_effect("rotate");
              call.perform();
            }
        - script.execute: delay_rot_enc

script:
  - id: delay_rot_enc
    mode: restart
    then:
      - delay: 700ms
      - light.turn_off: rgb_led

light:
  - platform: neopixelbus
    id: rgb_led
    name: "${devicename_upper} NeoPixel Light"
    type: GRB
    variant: WS2812
    pin: $ws_pin
    num_leds: $num_leds
    default_transition_length: 20ms
    restore_mode: ALWAYS_OFF
    on_turn_off:
      - lambda: |-
          id(effect_rotate_running) = false;
    effects:
      - addressable_lambda:
          name: "Rotate"
          update_interval: 50ms
          lambda: |-
            static boolean is_running = false;
            static uint32_t cnt = 0;
            static Color ring_colors[${num_leds}];
            static boolean rotate_right = false;
            static boolean rotate_on = false;
            static int rot_enc1_old = 0;
            
            if ( rot_enc1_old < int(id(rot_enc1).state) ){
              rotate_right = true;
              rotate_on = true;
            }
            else if ( rot_enc1_old > int(id(rot_enc1).state) ){
              rotate_right = false;
              rotate_on = true;
            }
            else {
              rotate_on = false;
            }
            rot_enc1_old = int(id(rot_enc1).state);
            
            if ( !id(effect_rotate_running)){
              id(effect_rotate_running) = true;
              it.all() = Color::BLACK;
              it[0] = Color(120, 0, 0, 0);
              it[1] = Color(174, 0, 0, 0);
              it[2] = Color(120, 0, 0, 0);
              return;
            }

            if ( !rotate_on ){
              return;
            }

            Color new_color;

            if (rotate_right){
              new_color = it[0].get();
              for (int i = 0; i < it.size() - 1; i++) {
                it[i] = it[i + 1].get();
              }
              it[it.size() - 1] = new_color;
            }
            else {
              new_color = it[it.size() - 1].get();
              for (int i = it.size() - 1; i > 0; i--) {
                it[i] = it[i - 1].get();
              }
              it[0] = new_color;
            }

```

zhekch, 2024-02-17:
Bumping this enhancement, as I really need it added

ssieb, 2024-02-17:
I have no idea what you're asking for.  You need to explain better.
Maybe https://esphome.io/components/light/#light-addressable-set-action is what you're looking for?

flaviut, 2024-02-21:
Some background. Let's say I have the following config:

```yaml
  - platform: template
    id: fan_speed_output
    type: float
    write_action:
      - lambda: |-
          for (int i = 6; i <= 9; i++) {
            id(status_light).addressable_set(i, 0, 0, 0, 0); // Turn off LED
          }
fan:
  - platform: speed
    name: "Filter Speed"
    output: fan_speed_output
    speed_count: 4
    id: my_fan
```

This does not work because `id(status_light)` returns [an `AddressableLightState`](https://esphome.io/api/classesphome_1_1light_1_1_addressable_light_state), which is exactly the same (no additional methods or fields) as `AddressableLight`: `error: 'class esphome::light::AddressableLightState' has no member named 'addressable_set'; did you mean 'AddressableLightState'?`

It would be nice if this worked, like the existing `id(light_1).toggle().perform();`. However, there's a decent workaround here:

```cpp
auto light = ((AddressableLight*)id(status_light).get_output());
for (int i = 6; i <= 9; i++) {
  light->get(i).set(Color(0,0,0));
}
light->schedule_show();
```