esphome/feature-requests #1669: Sensor filter: allow value if change is less than maximum change value

borpin, 2022-03-08:
**Describe the problem you have/What new integration you would like**

A filter (`max_change`) to remove 'spikes' in data by being able to define the maximum allowable change from the last good value to prevent rogue data.

I suggest it has a `number_of_values` configuration, which is the maximum number of values to be read that are outside of the allowable change range, before the new value is reported (a timeout). Allows for the possibility of sudden changes.

This is the reciprocal of the `delta` filter.

**Please describe your use case for this integration and alternatives you've tried:**

A room temperature sensor that occasionally outputs extreme variations incorrectly (e.g. last time 20 this time 1 next time 20.1) usually as the result of data corruption.

**Additional context**

micw, 2022-03-26:
You can workaround this by a lambda filter (pseudo code, not tested):

```
    filters:
      - lambda: |-
          if (abs(x-id(last_good_value))>100.0) return {};
          id(last_good_value) = x;
          return x;
```

Explaination:
* last_good_value is a global variable
* First check if the absolute difference is above a threshold. If so, return {} which skips this value
* If the delta is OK, store the value as last good value and return it

borpin, 2022-06-01:
Still be good to have as a built-in filter :)

micw, 2022-06-01:
You can define the variables within the lambda. They will be set on first call and be locally avauilable on all sucessive calls:

Example:
```
int counter = 0;
return ++counter;
```
Will return an increasing counter without the need of a global variable.

nagyrobi, 2022-06-01:
Problem is that after lambda exists with return, the local variables get lost. Need global variable, outside lambda, to keep track how many times that lambda has ran.

closed 2022-07-01 by nagyrobi

micw, 2022-07-01:
You are right, I forgott the "static" keyword. See https://esphome.io/guides/automations.html -> about persistent local variables.

No global needed with that ;)

borpin, 2022-07-02:
@nagyrobi hardly completed - there is still no suitable built in filter 😞

prchal, 2022-10-19:
Hi, I would vote for this filter.