esphome/feature-requests #1671: Support BLE Mesh

ThisIsTheOnlyUsernameAvailable, 2022-03-13:
**Describe the problem you have/What new integration you would like**
Native support of BLE mesh technology.

**Please describe your use case for this integration and alternatives you've tried:**
Meshing of all BLE devices so that range and signal is reliable.  I am not aware of an alternative in the ESPHOME ecosystem.

**Additional context**
There was an interesting discussion on the HA Facebook group around the increasing quality and variety of BLE devices available. What interested me was the potential to replace more expensive ZWAVE technology.

But the natural limitation of BLE is range.

Native support for BLE Mesh would be a fantastic addition to ESPHome as it would natively network all sensors together, and allow specific placement of an ESP32 to act as a repeater/amplifier to pass blackspots.

libby2cm, 2022-06-09:
I concur, this would be great.. i've been putting RGBCW downlights around the house and hate having them on wifi (not to mention tuya).. ble mesh control would be great (there are a few downlights out there that are ble only).

returntoreality, 2022-07-13:
@nagyrobi I do not think this is a duplicate of #1108. This ticket is about supporting Bluetooth LE mesh support, while #1108 is talking about WiFi meshing.

Dark-Noir, 2022-08-12:
Yeah, i also would love to see this ^^

Minims, 2022-11-15:
Would love to see ble mesh too to control my lights ! Thx

darfij, 2022-12-16:
would love to see this !!!

Minims, 2022-12-16:
There is a work in progress here : https://github.com/dominikberse/homeassistant-bluetooth-mesh

but i would prefer something running on esp32

Minims, 2023-02-20:
Is there anything planned for this ? 
I also see this for ESP32 & IDF : https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/esp-ble-mesh/ble-mesh-index.html

mdlima, 2023-10-30:
+1 to this. 🙂

ctomoiaga, 2024-05-24:
I’d also be very happy if this will be available

Bluscream, 2024-09-29:
I need this for https://www.youtube.com/watch?time_continue=2&v=kan2fQBgJLw&embeds_referring_euri=https%3A%2F%2Fwww.reddit.com%2F

Z3r10, 2024-10-03:
I second that! 👍

I have a bunch of those cheap Ledvance bulbs: [SMART+ Classic filament lamps with Bluetooth technology | LEDVANCE](https://www.ledvance.com/consumer/products/smart-home/smart-lamps/smart-bluetooth/smart-classic-filament-lamps-with-bluetooth-technology-c198383).

There is some effort done here-and-there on the HA Forum: [Add Ledvance Smart+ BT Bluetooth](https://community.home-assistant.io/t/add-ledvance-smart-bt-bluetooth/477478)

[Ultraworg](https://github.com/Ultraworg) made is own implementation, working but not perfect at the moment (thanks to him though!): [LEDVANCE_BLE_MESH](https://github.com/Ultraworg/LEDVANCE_BLE_MESH)

That would be awesome to have BLE Mesh support since those Bulbs are damn cheap (and the Mesh is itself a good thing for the house)

theopensourcerer, 2024-10-03:
> [Ultraworg](https://github.com/Ultraworg) made is own implementation, working but not perfect at the moment (thanks to him though!): [LEDVANCE_BLE_MESH](https://github.com/Ultraworg/LEDVANCE_BLE_MESH)

Interesting - Ultrawong has done what I was thinking about yesterday. I came up with the idea of replicating a Hafele "remote control" device - that supports on/off, brightness and colour temperature. 

Thanks for the link.

Minims, 2025-05-06:
Anything planned to include BLE mesh or not @esphome ?

mangaskahn, 2026-11-03:
Adding another voice to the group. Please add support for BLE Mesh.