esphome/feature-requests #1672: Support sleep mode for Grow Fingerprint sensor

manutenfruits, 2022-03-13:
**Describe the problem you have/What new integration you would like**

I want to be able to disable the Grow Fingerprint reader R502 when the ESP32 is put to sleep. This can be easily accomplished by always powering pin 6 (power for the fingerprint detector) and GND, and toggling ping 1 (VCC) with a transistor. This way, you can listen for finger detection and turn on pin 1 to communicate with the module, and turning it off after a timeout, or just before going to sleep.

**Please describe your use case for this integration and alternatives you've tried:**

The use case is for a battery-powered ESP32 fingerprint reader. This reader consumes 2uA of power with only the finger detection circuit enabled, but 30mA if the entire module is powered up.
As a workaround, I've tried to configure an [output](https://esphome.io/components/output/gpio.html) on a pin connected to an NPN transistor, and turn it on as early as possible using `on_boot` with priority 700, and off just before going to sleep. However, this setup does not work because there are only ~10ms between the transistor powered up and the first communication over UART that the sensor integration does. According to [the datasheet](https://cdn-shop.adafruit.com/product-files/4651/4651_R503+fingerprint+module+user+manual.pdf), it takes "<30 ms" to power up, but since the first call fails it cannot set up the component.
I have also tried using an automation delay after turning on the output in `on_boot`, but they work asynchronously and don't delay the components boot.

**Additional context**

I believe this should be integrated as part of the fingerprint module, with an extra `wakeup_pin` option to drive a transistor, and a `startup_delay` with a default value of 50ms or so to control how long to wait for the fingerprint to initialize.

alexborro, 2023-09-04:
I started playing with this fingerprint_grow integration for a R503 I bought.
I looked the code and it is quite easy to implement this feature, but there is a major problem. When the ESP32 enters in deep sleep, it takes around 10 to 15 seconds to connected to HA again when it wakes up. That is a no-go for a fingerprint reader. 

I also did some measurements and the ESP32 consumes around 50 to 80mA when running and the R503 less then 20mA. So yeah, you could same some power by turning off the reader, but that is just a fraction of the whole system usage. Is it worth ? Will it make much difference for a battery powered system ? I would not even consider ESP32+Wifi for a battery system.

manutenfruits, 2023-09-04:
There are some boards like the Firebeetle that seep only ~10uA in deep sleep, making this possible.

It is true that WiFi and connecting to HA takes a very long time, but my hope was to get this fixed along with some BLE communication feature or esp-now so it didn't need to connect to wifi to broadcast the error. If we get this fingerprint deep sleep functionality, paired with esp-now or BLE communication to another relay esp32, this is definitely doable.

grigi, 2023-09-25:
I found by setting my ESP32-C3 board to run at 40Mhz it's power usage drops to about 15mA whilst connected to Wifi.
I think one could argue that sleep more on the Grow could effectively half power usage in such a case, which would start to make it useful for battery backed device.

We are still talking of days and not weeks :shrug: 

This could open up the eventual battery usage for e.g. esp32-C6/H2 devices that support Zigbee for example. But Zigbee support seems to be a way out still #1397

grigi, 2023-09-27:
I setup esphome python venv on my notebook and manually tested those builds.

So nice to get builds within seconds as compared to within minutes on my slow HA box :sweat_smile:

alexborro, 2023-09-27:
> I totally failed to read your whole message, apologies for the confusion, but at least I got there in the end.
> 

Nice it worked. 
For anyone else, just use the branch I mentioned in my external component example (- source: github://alexborro/esphome@ext_comp) because it has a special change that allows the compilation of the component.

ferbulous, 2023-12-03:
Hi @alexborro  just wondering if using espnow like this component would also xtend the battery life?

https://github.com/afarago/esphome_component_bthome