esphome/feature-requests #1677: Add automatic/scheduled update feature to ESPHome

lukebaldan, 2022-03-17:
At present, if an update is available, the user has to manually click the 'update all' button on the ESPHome dashboard. It would be advantageous if this update operation could be scheduled automatically. I am running esphome on a slowish system, the compile times are long and my device count high so it would be nice to automate this operation to occur overnight.

jjansen85, 2022-03-21:
Was just thinking the same this weekend. Compilation is relatively slow on my Synology NAS, especially when it is a new ESPHome version. It would be nice if it can automatically compile at night for example. Actual update to ESP-device can be automatically as well or manually (but then it would be much quick as build files are already compiled).

nagyrobi, 2022-07-07:
Bad practice to do the updates in an unattended way imho. There could be breaking changes which could render some of your devices out of order, and restoring to previous state is not an easy task.

mikesalz, 2023-05-29:
> An alternative approach would be this: #1677

An alternative method would be this same thread?

I would love an option to auto-update my ESPHome devices.  Key word: Option. If someone feels comfortable with unattended updates, the risk should be up to the user.

Heisenberg2980, 2023-08-01:
This post includes an automation that can be setup to update all ESPHome devices at a desired time: https://community.home-assistant.io/t/auto-update-esphome-devices-via-automation/536573/28

VictorDrijkoningen, 2023-10-18:
i have built a github repo for a docker container which does do auto-updating the esphome devices. it is in a prototype state but is working. ([link](https://github.com/VictorDrijkoningen/auto-update-esphome))

VictorDrijkoningen, 2023-10-21:
> This post includes an automation that can be setup to update all ESPHome devices at a desired time: https://community.home-assistant.io/t/auto-update-esphome-devices-via-automation/536573/28

This is not possible with docker installed HA. This is only possible if ESPHome is installed as addon i believe

RudolfRendier, 2023-11-18:
It would be cool to have the binaries built on a schedule / on command by a call to an API.

VictorDrijkoningen, 2023-11-19:
@RudolfRendier  i created a tool to fix this problem. did not want to deal with periodically pressing the update button. 
[auto-update-esphome github repo](https://github.com/VictorDrijkoningen/auto-update-esphome/) 
[dockerhub link](https://hub.docker.com/r/victordrijkoningen/auto-update-esphome)