esphome/feature-requests #1678: Sending arbitrary Tuya MCU commands

Ceralor, 2022-03-17:
**Describe the problem you have/What new integration you would like**

I'm looking for a means of sending arbitrary TuyaMCU commands outside of the limited selection of Fan, Switch, and Number inputs.

**Please describe your use case for this integration and alternatives you've tried:**

A Tuya-MCU-based vacuum I own on an ESP32 and a Tuya-MCU-based smart lighting solution I own using an ESP8266 have a number of functions that can be controlled using the existing integrations. The light has most options easily controlled using the existing Tuya Dimmer integration, but it's a setup with each panel consisting of several LEDs, with configurable patterns that affect all panels simultaneously in the same way, able to be arranged arrange to make the patterns line up however one wishes. This requires sending the scene information to a datapoint as a long hex string of varying length based on the complexity of the scene, which seems to be a serialized JSON object as the color data is similar. Currently, I can't do that outside of some rather hacky lambdas that rely on using C++ calls I'm not very familiar with and feel unsafe using.

Similarly, the vacuum has most aspects outside of power controlled by text strings sent or received from various data points, with only battery being an int value and all others pretty much being bool or string. Even if I could only send specific commands based on services published to Home Assistant, utilizing the Template Vacuum integration to add it as opposed to an ESPHome-native Tuya Vacuum integration, it'd still work for me.

**Additional context**

Currently I'm using LocalTuya to control these devices, having written a Pyscript 'shim' of sorts for the effects for the light in the interim of not being able to flash them with ESPHome and control them fully local-only. I'd love to be able to take control of these through ESPHome instead, especially as it could give me better ways of controlling these in the future.

andreas-berg, 2022-11-21:
Late response but I had a similar requirement and found that the TuyaMCU already supports the requested feature. 
The [TuyaMCU-component](https://esphome.io/components/tuya.html) has public setter-functions for all the datatypes that Tuya-protocol uses. The datatypes are specified in the [Tuya component source](https://esphome.io/api/tuya_8h_source.html) as:
```c
enum class TuyaDatapointType : uint8_t {
   RAW = 0x00,      // variable length
   BOOLEAN = 0x01,  // 1 byte (0/1)
   INTEGER = 0x02,  // 4 byte
   STRING = 0x03,   // variable length
   ENUM = 0x04,     // 1 byte
   BITMASK = 0x05,  // 1/2/4 bytes
};
```
In my case I use types 1,2 & 4 and send commands with:

```c
void set_boolean_datapoint_value(uint8_t datapoint_id, bool value);
void set_integer_datapoint_value(uint8_t datapoint_id, uint32_t value);
void set_enum_datapoint_value(uint8_t datapoint_id, uint8_t value);
```
You should be able to send an arbitrary string value of type 3 with:
```c
void set_string_datapoint_value(uint8_t datapoint_id, const std::string &value);
```
Here's an example configuration from a Tuya-ceiling fan/-light, the switches send single Tuya-commands that trigger  sequences of commands in the TuyaMCU
```yaml
esphome:
...
uart:
  rx_pin: GPIO3
  tx_pin: GPIO1
  baud_rate: 9600

tuya:
# the 'tuya' device must have an ID so we can call it in the lambdas
  id: "tuyadevice"
...
switch:
  - platform: template
    name: Light - Dim to 30%
    turn_on_action:
#      - logger.log: "Dimming light to 30%"
      - lambda: |-
          id(tuyadevice).set_integer_datapoint_value(10,175);
...
  - platform: template
    name: Ceiling Fan Reverse 40%
    turn_on_action:
#      - logger.log: "Set Fan to REVERSE at 40%"
      - lambda: |-
          id(tuyadevice).set_boolean_datapoint_value(101,true);
...
#   fan speed: dpId = 3, type=4, range: 0x0=0 (minimal) - 0x06=6 (30%) - 0x0E=14 (50%) - 0x16=22 (70%) - 0x1F=31 (100%)
  - platform: template
    name: ${friendly_name} Ceiling Fan - Night mode (minimal speed)
    turn_on_action:
#      - logger.log: "$friendly_name Ceiling Fan Night mode"
      - lambda: |-
          id(tuyadevice).set_enum_datapoint_value(3,0x0);
```

If this answers the feature request then pls go ahead and close this issue.

jazzmonger, 2022-11-22:
would you guys know if is there any way to use the GPIO pins on modules like TYWE1S w/ TuyaMCU or some other method? there are several on my version (IO12 & IO14) that arent wired up or being used, but I can find no way to address them and use them like regular GPIO pins on std 8266 MCUs.

[image]

andreas-berg, 2022-11-22:
A bit off topic but sure, it should be possible. 
In my case I have a Tywe2s and only the power,gnd + serial tx&rx are used for communicating with the TuyaMCU. By soldering jumper wires to some free pins and configuring them in your Esphome (or Tasmota etc firmware), you could use them for wharever you want.

jazzmonger, 2022-11-22:
this isnt documented anywhere so I was unsure!  thanks for the info.

closed 2022-11-23 by ssieb