esphome/feature-requests #1680: Hostname customization

itec78, 2022-03-20:
**Describe the problem you have/What new integration you would like**

I'd like to customize wifi hostname. Now it's the same as device name, but it would be nice to choose a different one.

**Please describe your use case for this integration and alternatives you've tried:**

I have a lot of devices in my network configured with dhcp, and they appear on my router with the device name (eg: door)
I'd like to organize them with a prefix or with a custom hostname (eg: esphome-door or ha-door)
I tried changing device name but isn't a good solution.

**Additional context**

shmorgan, 2022-04-13:
I would agree with this request, right now I name the devices after the physical hardware and an index number, Makes keeping track of what hardware they are. It would be great if we could add a friendly name in the configuration, what would be displayed in the ESPHOME page and device Web Page.

nielsnl68, 2022-05-02:
i like this idea