esphome/feature-requests #1683: Add Ethernet support to `esp-idf`

nagyrobi, 2022-03-21:
**Describe the problem you have/What new integration you would like**
More and more Ethernet supporting boards are available these days.
- WT32-ETH01
- Olimex ESP32-POE/ISO, ESP32-EVB, ESP32-GATEWAY
- PRODINo ESP32 Ethernet
- TTGO T-Internet-POE ESP32
- wESP32 
- OpenHacks LAN8720
- ESP32-Ethernet-Kit

Currently ESPHome only supports Ethernet under `arduino` platform (config example below for WT32-ETH01):

```yaml
esphome:
  name: "Test"

esp32:
  board: esp32doit-devkit-v1
  framework:
    type: esp-idf

ethernet:
  type: LAN8720
  mdc_pin: GPIO23
  mdio_pin: GPIO18
  clk_mode: GPIO0_IN
  phy_addr: 1
  power_pin: GPIO16
  manual_ip:
    static_ip: 192.168.2.10
    gateway: 192.168.2.1
    subnet: 255.255.255.0
  use_address: 192.168.2.10
```
Results in error:

```yaml
ethernet: [source /config/esphome/test.yaml:38]
  
  This feature is only available with frameworks ['arduino'].
```

**Please describe your use case for this integration and alternatives you've tried:**
Please add Ethernet support to `esp-idf`. Less memory consumption.

**Additional context**
N/A

ma-lalonde, 2022-09-01:
1. How would one go about implementing this? Would it mostly be about copying the .h file and re-implementing the .cpp file?
2. How much work are we talking about, presumably?
3. What version of the IDF should be targeted? There are major changes in 5.0.0

nagyrobi, 2022-09-01:
Don't think it's just that easy, as far as I understand Ethernet is heavily relying on what's already in arduino base. For IDF wheel has to be reinvented.

tube0013, 2022-10-15:
Any type of timeline for when this could be implemented?

 I'm looking to try the hardware flow for UART from https://github.com/esphome/feature-requests/issues/1126#issuecomment-1010316131 but require Ethernet support (Olimex and WT32-ETH01)

nagyrobi, 2022-12-07:
Done in: https://github.com/esphome/esphome/pull/3564
Available from 2022.12 for LAN8720, RTL8201, DP83848, IP101 modules.