esphome/feature-requests #1685: Duty Cycle Sensor from a binary sensor

SeByDocKy, 2022-03-23:
**Describe the problem you have/What new integration you would like**
It would be a great addition to have a duty cycle sensor measuring activity from a binary sensor and not only from a GPIO pin

**Please describe your use case for this integration and alternatives you've tried:**

Example of usage:
```sensor:
  - platform: duty_cycle
    entity_id: id_of_my_binary_sensor
    name: Duty Cycle Sensor
```

nagyrobi, 2022-03-24:
similar to https://github.com/esphome/feature-requests/issues/1548
but reverse direction

SeByDocKy, 2022-03-25:
> similar to #1548 but reverse direction

I don't understand the relation between ? Can you explain ?

nagyrobi, 2022-03-28:
`slow_pwm` was originally also only able to operate with GPIO pins. @martgras extended that functionality to use other output plaftroms too.
What you want is the same, use not only GPIOs, but make it possible to use data from other platforms (like you mentioned `binary_sensor`, but could be of course any other presenting binary states like `switch`, `light` etc.
Direction is reverse, the updated `slow_pwm` generates data towards platforms (sends triggers to them), what you want is to receive information or triggers from platforms.