esphome/feature-requests #1688: ESPHome connecting to HA over the internet from remote location

Pro-Control, 2022-03-25:
**Describe the problem you have/What new integration you would like**

**Please describe your use case for this integration and alternatives you've tried:**

**Additional context**

nagyrobi, 2022-03-26:
Wouldn't do this on an open public internet connection...

Use some cloud based stuff like Tuya etc.

Or use a small openwrt-based mini-router to call in to your home site via a secured vpn, and then pipe the api over that.

Pro-Control, 2022-03-27:
I hope to find a way to do it like as an add-on in HA work as a host server for external ESPHome device 
@nagyrobi  thanks for your advice I know these methods

droscy, 2023-07-05:
You could test with wireguard PR https://github.com/esphome/esphome/pull/4256 installing wireguard in your HA.

h3ndrik, 2023-08-28:
I'd like to request this too. No idea why we shouldn't do this via the internet. I have sensors at my parent's home, would like something at my desk at work that'd connect home via the guest network etc. Maybe give some device to the neighbours. There are several valid use-cases. And secure and encrypted connections over the internet are well studied.

Currently that is possible via MQTT. But IPv6 support isn't there yet and everything is just a PITA.

droscy, 2025-06-20:
Don't you like WireGuard?
Instead of exposing HA API endpoint you expose WG port from HA side, then ESPHome create the WG link and you can reach the device through that link.