esphome/feature-requests #1690: Hbridge switch or output platform to drive solenoids or DC motor

raidnet-ms, 2022-04-02:
**Describe the problem you have/What new integration you would like**

I want to drive a bistable solenoid or a motor without PWM feature or anything else with a binary output (no speed control needed).
**Please describe your use case for this integration and alternatives you've tried:**
I need to drive an H-Bridge module. Unfortunately the Fan Hbridge components seem to work only with a float value needed for the speed so I cannot use to drive a bistable solenoid (wich actually works as a DC motor to move in or another direction).

**Additional context**

H-Bridge fan platform
https://esphome.io/components/fan/hbridge.html

twanjaarsveld, 2022-08-22:
Did you find any solution 
i'm looking to salvage my old tuya doorlock and controll the lock using an esp8266 but it just needs 1 second of voltage to lock or unlock

dwmw2, 2024-09-08:
https://github.com/esphome/esphome/pull/7421 should do it?

raidnet-ms, 2024-09-08:
> https://github.com/esphome/esphome/pull/7421 should do it?

Oh yes please. Wish I was enough skilled to do that by myself.

Perhaps I could contribute with a couple of beers!
😂

printplaatreparatie, 2024-12-18:
The naming is confusing to me. I use H-bridges in my hardware that are not latching, as well as H-bridges made from FETs.

It looks like esphome/esphome#7421 can be used on any bistable relay. Wouldn't it be better to address the H-bridge in documentation if the name H-bridge switch is only used to make it easier to find?

See [https://github.com/esphome/esphome/pull/7421#issuecomment-2336763365](https://github.com/esphome/esphome/pull/7421#issuecomment-2336763365)

dwmw2, 2024-12-18:
The current documentation will show up when you search for "latching" but not "bistable". Want to submit a docs PR to fix that?

[image]

raidnet-ms, 2024-12-18:
You finally got your dopamine @dwmw2 . Now you gonna take my beers!! 😁