esphome/feature-requests #1693: Make a Total energy sensor that is restored after reboot

niklerus, 2022-04-05:
**Describe the problem you have/What new integration you would like**
There's a "Total Daily Energy" sensor that persists through reboots but it would be even more usable if there was a "Total energy" sensor that was persistant as well.

**Please describe your use case for this integration and alternatives you've tried:**
There's a Power Consumption application (https://github.com/klaasnicolaas/home-assistant-glow ) that works great but if you reboot the ESP32 the total energy number is lost but the daily is preserved. Would make sense to be able to preserve the total as well. (probably more useful than saving the daily consumption)

**Additional context**

dz0ny, 2022-04-11:
This can be done via https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/memory-types.html#noinit-dram

```yaml

  on_boot:
    priority: 800.0
    then:
    - lambda: |-
        __NOINIT_ATTR uint32_t my_data;
        id(some_value_sensor).publish_state(my_data);
sensor:
    ...
    id: some_value_sensor
    on_value:
        then:
        - lambda: |-
            __NOINIT_ATTR uint32_t my_data;
           my_data = x;
```

jumping2000, 2022-07-11:
the total energy sensor would be very useful 👍

Bert-R, 2022-11-22:
It would indeed be useful if this would be available out of the box. For as long as it isn't, you can build it yourself, rather easily. I made a PR on the Athom smart plug configuration that does it: athom-tech/athom-configs#22

Roving-Ronin, 2024-04-25:
+1 should have a 'Total Energy Daily' (as exists now) and a 'Total Energy' (i.e. lifetime / at least time between firmware updates of device). Bert-R's on athom v2 plug works great.

nkinnan, 2024-09-09:
This is now natively supported, I added a section to the doc explaining how to use it: https://esphome.io/components/sensor/total_daily_energy.html#lifetime-instead-of-daily

Roving-Ronin, 2024-09-10:
This only shows the Daily Energy, based upon @dentra's work, I use the following that provides a breakdown for Today / Yesterday / Week / Month / Year usage.

https://github.com/Roving-Ronin/myHomeAssistant/tree/main/components/energy_statistics

Example of its usage in Athom plugs, is this module that is loaded by the 'overall' yaml for those type plugs: https://github.com/Roving-Ronin/myHomeAssistant/blob/main/esphome/sensors/athom-power-plugs.yaml

In webserver v3 interface, it looks like:

[image]

Note: 'NA' is not an error for year, the sensors start showing data come the next Monday / 1st of Month / 1st Jan etc.