esphome/feature-requests #1695: Extend the UPDATE ALL button with UPDATE ONLINE, and UPDATE AVAILABLE function

Korte68, 2022-04-06:
It would be great if we can update with one click the following devices:

- ALL (it is already exist)
- ALL ONLINE (update all online ESP)
- ALL OLDVER (update only the ESPs which status is UPDATE AVAILABLE)
- ALL ONLINE and OLDVER (update only the ESPs which status is ONLINE and UPDATE AVAILABLE)
- ALL SELECTED (We should able to select as many ESP as we want, and update them 1 one click)

**Please describe your use case for this integration and alternatives you've tried:**

**Additional context**

tungmeister, 2022-04-17:
I'd love this as well, I have over 50 epshome devices so updating them all takes a huge amount of time. I'd love the option to select a specific set to update or to only update devices with updates pending. The "update available" message on devices also seems to only show when the update hasn't been compiled. Once complied the indicator goes away regardless of if the upload has been succesful.

chatziko, 2022-05-08:
This would be hugely useful indeed, for those having 20+ devices. And seems like a small change from the current behaviour. One could write a script, of course, but the dashboard is super easy to use.

cloom, 2022-06-03:
Hello,

Great FR. It could be nice to select but also create groups. So AVAILABLE, OLDVER etc would be auto-generated groups but we could also group by type, room or whatever the user needs.

My use case would be my ESP homes with custom light effects, when I modify the light effects I could just update all of them in 1 click :)