esphome/feature-requests #1697: Add support for VL53L1X Tine of Flight sensor

Scoff123, 2022-04-07:
**Describe the problem you have/What new integration you would like**

I would like to see support for the ST VL53L1X Tine of Flight sensor within ESPHome. Currently this has to be implemented via a custom component.

**Please describe your use case for this integration and alternatives you've tried:**

This is to be used alongside a Lolin D1 Mini Pro board in order to measure fluid level in an oil tank. 
There is support for the ST VL53L0X Time of Flight sensor, however this sensor uses 8 bit I2C registers. The VL53L1X uses 16 bit I2C registers. References to previous requests for this sensor and 16 bit address support:

https://github.com/esphome/feature-requests/issues/836

https://github.com/esphome/feature-requests/issues/875

https://github.com/esphome/feature-requests/issues/1001

**Additional context**

The latest 2 releases of the ESPHome addon for Hone Assistant no longer compile the custom component correctly. Details of compile error are here:

https://github.com/esphome/issues/issues/3203

mrtoy-me, 2023-06-26:
If there is still any interest, I have done a  ESPHome VL53L1X custom component which you are welcome to test.
It uses esphome I2C base level functions to create 16bit address I2C functions that work for VL53L1X. Not all functions implemented so far but you can set distance mode and read distance and range status.

frederickjh, 2024-07-05:
The abstraction for reading and writing to 16 bit I2C registers was added to esphome/esphome#4844 and merged. #875 was closed as a result of this. Hopefully this paves the way  for support for the VL53L1x to be added.

bojanpotocnik, 2024-11-05:
Shall we close this as a duplicate of https://github.com/esphome/feature-requests/issues/836 and provide any working implementations there, to eventually merge this feature?