esphome/feature-requests #1699: Add support round display: GC9A01 240x240 IPS

error105, 2022-04-10:
**Describe the problem you have/What new integration you would like**
Add support for beautiful round display with GC9A01 driver.

**Please describe your use case for this integration and alternatives you've tried:**
Support for round charts.

**Additional context**
Documentation from Waveshare:
https://www.waveshare.com/1.28inch-lcd-module.htm

nickwilkie, 2022-06-16:
I second this! Would love to use this screen, particularly if the example renders like the 'gauge' could be included as a parameter rather than needing to draw it manually.

jlips24, 2022-06-28:
This would be an amazing addition for devices like temperature/humidity sensors!

Nilaik, 2022-07-16:
Just found this pull request: https://github.com/esphome/esphome/pull/3625

pvh0, 2022-09-29:
> Just found this pull request: [esphome/esphome#3625](https://github.com/esphome/esphome/pull/3625)

Got this working with:

external_components:
  - source: github://pr#3625
    components: [ gc9a01 ]

# Example config.yaml
spi:
  mosi_pin: GPIO13
  clk_pin: GPIO14

display:
- platform: gc9a01
  reset_pin: GPIO26
  cs_pin: GPIO15
  dc_pin: GPIO33
  rotation: 180
  lambda: |-
      it.line(0, 0, 100, 50);

gattrill, 2023-02-17:
Thanks for this, I bought a few of these ages ago but never got around to doing anything and was pleased to see all the hard work done.  They are *really* nice and very very cheap, I assume they are surplus/rejected smartwatch displays.

Just wondering if there are any components/packages/examples for drawing a gauge on a display with ESPHome? I haven't done much with displays yet other than drawing text, and I'm sure it's a common thing for people to do.

error105, 2023-02-17:
I make gauges on it, but in PlatformIO and just read values from mqtt from HA. Thats simple ;)

gattrill, 2023-02-21:
> 

Yeah I want to keep everything in ESPHome really. I'm just writing a component so you can draw a nice needle gauge from a sensor, based on the graph component.

error105, 2023-02-21:
Can u show ?

KevynWithaY, 2023-08-13:
I'd like to see the needle guard drawing component also

smkent, 2024-02-17:
[I combined the various PRs I found here, added a color palette fix, and added some performance improvements](https://github.com/smkent/esphome/pull/3) to the `gc9a01` component on [my fork](https://github.com/smkent/esphome).

[image]

clowrey, 2024-03-12:
Support is going mainline! https://github.com/esphome/esphome/pull/6351

TheOriginal92, 2024-04-01:
Is there any tool to create a nice gui with his display?

nagyrobi, 2024-04-02:
https://github.com/esphome/esphome/pull/6363

closed 2024-04-15 by clydebarrow