esphome/feature-requests #1700: Add IR Remote Climate support for Dometic FreshJet 3000

DoctorMarshall, 2022-04-10:
**Describe the problem you have/What new integration you would like**

It would be great if Dometic FreshJet was integrated in IR Remote Climate.

**Please describe your use case for this integration and alternatives you've tried:**

I will be setting up the AC unit with regular IR commands but would be great to integrate it into IR Remote Climate. Will update this issue once headway is made.

**Additional context**

BreitiDE, 2024-02-18:
any progress or custom solution available ?

nagyrobi, 2024-06-27:
Did you try the `zhlt01` climate or protocol, or the `zhjg01` protocol from [IR Remote Climate](https://esphome.io/components/climate/climate_ir)? These are remotes used with many locally branded air conditioners.

prebsit, 2026-09-27:
For anyone landing here with a newer FreshJet (FJX4/FJX7, or the FJZ, the ones that work with the Dometic Climate app over Bluetooth): those can be controlled over Bluetooth with an ESP32, with no IR needed and with state coming back from the unit. That's here: https://github.com/prebsit/dometic-fjx7-ha. It won't help the older IR-only FreshJet 3000 though....sorry :(