esphome/feature-requests #1702: Adafruit MPRLS Ported Pressure for Radon Vacuum Monitoring

switchtower, 2022-04-12:
**Describe the problem you have/What new integration you would like**

I would like to use the I2C MPRLS sensor from Adafruit: https://www.adafruit.com/product/3965

https://github.com/adafruit/Adafruit_MPRLS

**Please describe your use case for this integration and alternatives you've tried:**

I would like to use this sensor to measure the vacuum from a radon ran to verify the fan is working.

**Additional context**

kj4dxk, 2022-12-06:
I would upvote this.

thegreatco, 2023-06-27:
I'll 3rd this

dwijn, 2023-11-29:
FYI the Honeywell MPRLS sensors work over I2C using the [honeywellabp2_i2c](https://esphome.io/components/sensor/honeywellabp2_i2c.html) sensor platform. Here's an example YAML snippet:
```
sensor:
  - platform: honeywellabp2_i2c
    pressure:
      name: "MPRLS Pressure"
      unit_of_measurement: "PSI"
      min_pressure: 0
      max_pressure: 25
      transfer_function: "A"
    address: 0x18
    update_interval: 60s
```