esphome/feature-requests #1703: add a "rename" action in the dropdown menu of a node, in home-assistant's esphome dashboard

Ironirc, 2022-04-14:
**Describe the problem you have/What new integration you would like**

I tried different ways, resulting different sorts of connection loss.

**Please describe your use case for this integration and alternatives you've tried:**

**Additional context**

nagyrobi, 2022-04-15:
There's a small pen icon in the device settings, near _Device info_ - just for that.

Ironirc, 2022-04-15:
We 're clearly talking about something different?
[image]

nagyrobi, 2022-04-15:
Ah there. Just rename the yaml file.

nielsnl68, 2022-05-02:
I like to have this option as well. 
Yes i know you can do that but it would be cleaner when the dashboard take the action. 
It takes 2 compile sessies and uploads om a device to rename. 
fist with the changed name and wifi: use_address: old_address.
and a second time where we removed the use_address again.

My suggestion would be that the filename is the current name used for mDNS resolve.
when the esphome name is changed after the file is uploaded correctly the filename is changed to the new name.

x0001346, 2022-05-12:
This would be a really useful feature.
In my case, so far I have 26 devices and due to a bad naming scheme I used I need to change the name of most of them.
I have already changed the name of some of them but this is prone to error and is easy to lose the connection with the device, this is specially con concerning since I'm not able to physically access the device.

pat1300, 2022-05-12:
I would also love this feature :-)

glmnet, 2022-06-18:
Done in 2022.5.0 https://esphome.io/changelog/2022.5.0.html

closed 2022-06-18 by glmnet