esphome/feature-requests #1705: Support for Adafruit ESP32-S2 TFT Feather

GitHubGoody, 2022-04-16:
**Describe the problem you have/What new integration you would like**
Support for the Adafruit ESP32-S2 TFT Feather (https://learn.adafruit.com/adafruit-esp32-s2-tft-feather)

**Please describe your use case for this integration and alternatives you've tried:**
I'm trying to get this board to work with HA using ESPHome. I get various errors as documented here (https://community.home-assistant.io/t/first-esphome-project-esp32-s2-feather-board-stuck-on-initial-configuration-install/412097/8)

**Additional context**
No

GitHubGoody, 2022-05-05:
This board is now listed in the PlatformIO registry.

https://registry.platformio.org/platforms/platformio/espressif32/boards

nagyrobi, 2022-07-01:
https://github.com/esphome/feature-requests/issues/996
https://github.com/esphome/esphome/pull/3264