esphome/feature-requests #1712: Please add Software UART to ESP32

it-guy92, 2022-04-26:
**Describe the problem you have/What new integration you would like**

**Please describe your use case for this integration and alternatives you've tried:**

**Additional context**

ShunichiSan, 2022-07-13:
I've encountered the same problem. The doc says "When the hardware UARTs are all occupied, ESPHome will fall back to a software implementation that may not be accurate at higher baud rates." 
I'm migrating  a project from an Arduino nano which easily handled 4 simultaneous DFplayers using softwareserial (each requires its own serial connection). However with ESPHome on an ESP32 only two will work regardless of my pin selection. The third UART is presumably being used by the logger.  Each DFPlayer works if I initiate them alone, but enabling more than two only results in the first two working. It would be great if the software UARTs were enabled, or at least update the docs.

wawibu, 2022-08-06:
Glad to found this as I run into the same issue. I want to use an ESP32 with more than 3 UART and I also understood the documentation in the way that EspHome would fall back to sw uart. 
Using the hw uart already for GPS, MPPT charger and SDM120. Now I want to add the Nextion Display to it, but didn't get it working as it didn't fall back to sw uart if all hw uart already occupied.
Is there any option to force to sw uart?

lwqcz, 2022-08-30:
I found this repo and it seems to be working ... https://github.com/KinDR007/VictronMPPT-ESPHOME

rnizametdinov, 2022-09-05:
> I found this repo and it seems to be working ... https://github.com/KinDR007/VictronMPPT-ESPHOME

i think it use HW serial on ESP32. 
I dont found any custom softserial code in this repo.

simbaja, 2022-10-17:
Ran into this issue as well.  ESP32-S2 only has two UARTs apparently, so it's even more problematic.  It caused me a bit of confusion, since the documentation says it'll fallback to software, but nothing in the code seems to allow it to do so.

formatBCE, 2023-03-05:
Ping on this. I have 3 UARTs configured, and whichever is written last, doesn't work.
Should i use GPIO1 and GPIO3 for third device, to share it with logger?

ssieb, 2023-03-05:
You can't share it, you would have to disable serial logging.  All ESP32 variants have 3 hardware uarts, but esphome will currently only use uart0 if you specify GPIO1 and 3 for the pins.

formatBCE, 2023-03-05:
> You can't share it, you would have to disable serial logging.  All ESP32 variants have 3 hardware uarts, but esphome will currently only use uart0 if you specify GPIO1 and 3 for the pins.

Yes, already realized it. Still, no software UART despite having it in docs...

ShunichiSan, 2023-03-05:
I think they just need to update the docs since there's no intention at this point to add software UART.

ssieb, 2023-03-06:
The note mentions it only for 8266, but I suppose the "Hardware UARTs" section could clarify that the software option only applies to 8266.

Fabian-Schmidt, 2023-03-20:
FYI: Victron has documented their Bluetooth advertising protocol so you could switch from UART to Bluetooth for Victron devices. This will not solve the Software UART request, but  I am happy with a BLE link and therefore need two UART less on my ESP.

<https://github.com/Fabian-Schmidt/esphome-victron_ble>

nagyrobi, 2023-03-20:
> The note mentions it only for 8266, but I suppose the "Hardware UARTs" section could clarify that the software option only applies to 8266.

Done: https://github.com/esphome/esphome-docs/pull/2734

jhollowe, 2023-09-19:
As a note: if you set the logger baud rate to 0, it will not use up a hardware UART.

e.g.
```yaml
logger:
  level: DEBUG
  # disable putting logging on the HW UART
  baud_rate: 0
```

vekexasia, 2023-12-23:
@jhollowe are you able to see the logs in wifi with logger level debug and baud_rate 0?

ssieb, 2023-12-23:
@vekexasia yes, it only stops the logger sending data on the uart.

JoHo94, 2024-01-21:
+1 for this feature. I want to use one esp to control multiple UARTS. So it would be awesome to use the 3 Hardware UARTs + some Software ones.

bzumik1, 2024-03-24:
I would also love to see this feature implemented! 🙂

zechnkaas, 2024-05-21:
+1 here (also like to add 4 Victron Devices and one DALY BMS)

did anyone get in the meantime try on of those to expand UART

[https://esphome.io/components/weikai.html](url)