esphome/feature-requests #1714: Ability to create multiple HA devices from one ESPHome config

Ulrar, 2022-04-29:
**Describe the problem you have/What new integration you would like**
Currently it seems one esp board shows up as one device in home assistant, which in turn owns the entities esphome has defined.
I'd like to be able to create multiple devices, and assign each switch / sensor or any other entities to specific HA devices.

**Please describe your use case for this integration and alternatives you've tried:**
I'm using an esp32 as a ble relay for switchbot devices, and all the entities I define in esphome show up under one device in HA, which is a bit messy.
It'd be much cleaner IMHO if esphome could expose one device per switchbot, which could all have their own battery levels, identify buttons and whatever entities that specific device happens to expose.

Currently the only way to do this is to use one esp per switchbot, which would be a lot of esps.

**Additional context**
Apologies if this already exists or has already been requested, I did try searching but `device` is a pretty generic term which returns pages and pages of results on this repo. Didn't see anything that sounded like this when skimming through, may have missed it.

apophisnow, 2022-11-18:
I need this as well. I have a LilyGo 8 relay board which uses one esp32 to control the relays. I would like to have 8 devices in HA, one for each relay which I can add numbers to for controlling amount of liquid to be dispensed individually for each pump.

Ulrar, 2022-11-18:
Now that the bluetooth proxy is a thing and actually works well I expect this to be a lot less needed, but it'd still be a useful addition IMHO.

zacken, 2022-11-19:
This would be a great feature that would enable creating gateways with esphome. Lets say you have a proprietary serial interface that or bus that lets you read and write to multiple devices.

randybb, 2022-11-19:
I don't see any reason for such feature and it is more HA feature than ESPHome feature. A device in HA represents source from which entities are coming from. In many cases only entity is the thing you need to know and it can be on any device, for example you can easily migrate a temp sensor from one device to other one with preserving its name. Any logical separation under device is completely useless, If you need to have a nice view - just create a card in HA dashboard. If you want to have it automated, just use this https://github.com/thomasloven/lovelace-auto-entities and use the same naming convention for all your entities.

apophisnow, 2022-11-19:
Yeah, after talking more about it with jessierockz, I agree. It wouldn't really make sense to implement, and like you say is really more of a feature request for home assistant anyway. Should probably just close this feature request as will not implement.

closed 2022-11-19 by nagyrobi