esphome/feature-requests #1715: Support for ILI9488 display controller

FloFri, 2022-04-30:
**Describe the problem you have/What new integration you would like**

The ILI9488 display controller is used in some "larger" TFT displays (https://www.aliexpress.com/item/1005003773199628.html). It is similar to the ILI9341 display controller but needs some changes to work:
* Changing the resolution to 480*320
* Changing the colors to 888-values (e.g.: 24 bit rgb)

**Please describe your use case for this integration and alternatives you've tried:**

I already tried to use the ILI9341 component but it did not work (display is flashing white). I also tried to implement the needed changes myself but had no success. Changing the resolution resulted in a boot loop and only changing the way colors are handled did not work either (maybe my math was wrong).

**Additional context**

An arduino library for ILI9488 can be found here: https://github.com/jaretburkett/ILI9488

Zoogara, 2022-05-07:
Plus 1, but this has been requested before with no  success. We can only hope.

zala143, 2022-05-07:
plus 1 - support for ESP32 TouchDown that includes the ILI9488 display

hyq023488, 2022-05-15:
支持 ESP32 触控，包括 ILI9488 显示屏

workingmanrob, 2022-05-19:
I am currently using an ili9486 based controller with an esp32 using a closed PR from MasterTim17.

https://github.com/esphome/esphome/pull/1523

Not sure if this helps at all but it does work just fine.

markusressel, 2022-06-23:
Would like to see support for this as well, because it is used in the ESP32 TouchDown (aka FreeTouchDeck).

hoodlumj3, 2022-09-16:
A big plus 1 from me - would really like this to be supported with touch XPT2046 - have a few lying around atm from my freetouchdeck days. had the idea having these on walls and desks integrated with home assistant.

@Zoogara what was the deal on prev request and being denied support?

Zoogara, 2022-09-17:
> @Zoogara what was the deal on prev request and being denied support?

Hmm - that was a while back - I should have linked the issue number.

The fact that this is still open is a good sign.  Also that MasterTim17  has supposedly got is working in a forked version of ESPhome: https://github.com/esphome/esphome/pull/1523

I am currently pairing some spare freetouchdeck adapter boards from Dustin Watts with the 3.2" ILI9341 - the displays are pin-compatible and although lower resolution are only slightly smaller.

workingmanrob, 2022-09-17:
I have it working right now.  It's not perfect but it's functional.  I mean the colours are a bit of a disaster but I didn't care for my purpose.

I used https://esphome.io/guides/faq.html#how-can-i-test-a-pull-request like this.

```
external_components:
  - source: github://pr#1523
    components:
      - display
      - ili9486
```

I also opened a feature request so :+1: it all day - https://github.com/esphome/feature-requests/issues/1853

nielsnl68, 2022-10-05:
I just created a PR esphome/esphome#3795 that does the same thing,

BerkshireBugsy, 2022-10-15:
Plus 1 from me too! I have a 480x320 SPI display I bought from AliExpress which I can't get working despite using different types of drivers (I mainly focused on the ILI9341 display because - in my ignorance - I thought that I was getting.

This is such a good display model - at a nice price - I would love to see it working with ESPHome :)

I've tried github://pr#1523 (as above) which only shows a full screen white fill. In the original 9341 (see photo) tests I got a partial display - before I found out the display was probably a 9488.
[image]

nielsnl68, 2022-10-15:
I added the ili9488 to the ili9xxx component only to find out that the internal memory was not enough. We need to find a way to resolve that before it going to get drivers like this one to work within esphome.

But please know, i am working on it.

BerkshireBugsy, 2022-10-15:
Thank you very much for the update

JeroenNL85, 2022-11-03:
@nielsnl68 

Thank you for your work, 
I have a ILI9488 and tried your PR, but got the same error.

[C][spi:023]: Setting up SPI bus...
[D][ledc.output:041]: Calculating resolution bit-depth for frequency 1000.000000
[D][ledc.output:046]: Resolution calculated as 16
[E][display:022]: Could not allocate buffer for display!

If there is a fix, i would love to test it.

nielsnl68, 2022-11-04:
You need to use an esp32 with **PSRAM**

Wummeke, 2022-11-17:
@nielsnl68 I am using your PR to get my ILI9488 screen working. I noticed that lines 35 - 37 in display.py (where the models are declared) are commented. Is that on purpose, or can I safely uncomment line 37?

nielsnl68, 2022-11-17:
when you sure it works for you (You have a esp32 with PSRAM) then you can safely uncomment the ILI9488.

Wummeke, 2022-11-17:
I do, since today. Unfortunately, I don't get the screen to work yet. I only get a blank screen with a grey-ish section on the right (when I hold the screen in landscape, wit pins to the left).

nielsnl68, 2022-11-17:
that sounds as if not the whole screen is initialized. Are you on Discord as well. When so could you ping me there. then we can walk trough the config and see what to fix.

carryonrewardless, 2023-05-03:
I have this display https://www.amazon.com.au/Touch-Screen-Module-Resolution-Arduino-Mega2560/dp/B07F2XM5HW
but have not been successful.  Would love it to work on ESPHome.
Controller: ili9481 ili9468, ili9488 hx8357, or r61581
Pin info :
 1	 NC	 
 2	 VDD	 
 3	 VDD	 
 4	 CS	 
 5	 RS	 
 6	 WR	 
 7	 RD	 
 8	 RESET
Has anyone got this working?

nielsnl68, 2023-05-03:
> Has anyone got this working? How much RAM is required to run this one?

You need an esp32 with 2MB or more of PSRAM. You can calculate the memory size multiply the height, width and then by 2.
For your display that is around 300KB.

carryonrewardless, 2023-05-03:
Being pin compatible with Arduino Uno I paired it up with this https://99tech.com.au/product/esp32-uno-d1/ and have it working using Adafruit_GFX.h and MCUFRIEND_kbv.h libraries and mqtt to send it data for display. It's pretty basic and rather ugly.  I want to have another attempt to get it working with ESPHome.
The ESP32 Uno D1 seems like it should have enough memory but its above my pay grade.
Any comments?

carryonrewardless, 2023-06-04:
Ah I found this for the display. http://www.lcdwiki.com/3.5inch_Arduino_Display-UNO
It states it as a  ILI9486/ST7796