esphome/feature-requests #1721: OpenHASP as ESPHome module

nagyrobi, 2022-05-05:
**Describe the problem you have/What new integration you would like**
There's OpenHASP: https://openhasp.haswitchplate.com making possible nice stuff like:
[image]

There's the new [LCD menu](https://github.com/esphome/esphome/pull/3406) PR implementing the concept of associating menu elements to esphome components (currently _select_, _number_ and _switch_).
This concept could be also extended over OpenHASP. 

OpenHASP creates drawable [objects](https://openhasp.haswitchplate.com/0.6.2/design/objects/) based on LVGL library, these have various properties but the most important part is that these can also get/set values in similar ways. ESPHome's _select_, _number_ and _switch_ can be easily matched to OpenHASP's drawable objects:

| ESPHome component | OpenHASP object types |
| ---- | ---- | 
| switch |  binary and toggle types
| number |  range types
| select |  selector types

OpenHASP also has a [gropid](https://openhasp.haswitchplate.com/0.6.2/configuration/gpio/#group) property which can be set for each object, intended to link drawn objects values with GPIO-related peripherals. So theoretically this is compatible with the concept above, ESPHome's components have to be put in OpenHASP's groups to link them together.

**Please describe your use case for this integration and alternatives you've tried:**
ESPHome and OpenHASP advantages could be combined into one:
- ESPHome has exceptional Home Assistant integration support and can handle a plethora of periplerals
- OpenHASP offers an exceptional implementation of LVGL to make interactive user interfaces

No need to reinvent the wheel on any of the sides. Currently ESPHome lacks of such a possibility, and OpenHASP lacks a way to easily integrate external peripherals locally.

**Additional context**
OpenHASP's author @fvanroie made a few tests:
https://github.com/fvanroie/esphome-lvgl
https://www.youtube.com/watch?v=9A8NO0gWaAI

fvanroie, 2022-09-11:
Take a look at the https://github.com/nielsnl68/SwitchPlate project which aims to integrate LVGL into ESPhome.

nagyrobi, 2024-04-02:
Work in progress, LVGL will be natively supported by ESPHome: https://github.com/esphome/esphome/pull/6363

[image]

nagyrobi, 2024-08-30:
Done: https://esphome.io/components/lvgl/

closed 2024-08-30 by nagyrobi