esphome/feature-requests #1725: Modbus component doesn't support user-defined functions

gazoodle, 2022-05-11:
**Describe the problem you have/What new integration you would like**

I would like the modbus component to forward response bytes from modbus user-defined functions so that my custom component can receive the response and process it.

**Please describe your use case for this integration and alternatives you've tried:**

I have a Century Variable Speed pump that supports a [user-defined set of modbus commands](https://github.com/gazoodle/CenturyVSPump/blob/main/docs/Gen3%20EPC%20Modbus%20Communication%20Protocol%20_Rev4.17.pdf).

If you try to use the `modbus_controller` and various sensors and/or switches, the response from the pump is never received since it doesn't conform to the publicly documented pattern. Modbus supports a range of [user-defined functions ](https://modbus.org/docs/Modbus_Application_Protocol_V1_1b3.pdf) (Chapter 5).

**Additional context**

I have a small patch that will allow the modbus component to forward the response bytes to a new custom class that implements the pump protocol which will allow my [custom component](https://github.com/gazoodle/CenturyVSPump) to function correctly

I'll create the PR when I've got a number to create a branch against.

martgras, 2022-05-11:
I'm going to review your PR but I was wondering why https://esphome.io/components/sensor/modbus_controller.html#using-custom-command  doesn't work for you (maybe modbus_controller should get an update as well if it is lacking a feature)

gazoodle, 2022-05-12:
I did originally try to use the custom-command, but the response lambda never got called until I made the changes in this PR.

I then continued into `modbus_controller` but IIRC there seemed to be assumptions about register and count which I wasn't sure about at the time.

EDIT: Just retried using the controller and a custom sensor and while it now works (i.e. the command is sent and the response is dispatched), the logging is misleading, you get things like **`Sending next modbus command to device 21 register 0xD28E count 1`** which doesn't really make sense in this context. I'm sure we could resolve that.

There are a couple of other things that might be necessary, such as when the pump detects an error condition, it helpfully sets the high bit on the function code in the reply (as per modbus protocol), but the next byte is set to it's internal NACK code which is lost since the error reply is never forwarded to the sensor (or whatever), and of course error handling sometimes needs to have a catch-all system (for logging mainly) which is not nice in the yaml lambda syntax.

It would be better to use your `modbus_controller` layer, and I think I'll head toward that in a future iteration.