esphome/feature-requests #1729: Support for Sonoff MS01

stskrp, 2022-05-16:
**Describe the problem you have/What new integration you would like**
Support for Sonoff MS01 moisture sensor

**Please describe your use case for this integration and alternatives you've tried:**
Sonoff TH10/16 now has support for new sensor - MS01 https://sonoff.tech/product/accessories/ms01/
It is already supported in Tasmota https://github.com/arendst/Tasmota/pull/15335/files
But not sure how to port it to esphome.
Would be great to have support for MS01 is esphome too.

**Additional context**

kahhu3, 2022-07-19:
I'm using the Tasmota  software successfully but would appreciate support in ESP home as well.

Zebble, 2022-10-16:
Trying to add this through [PR#3901 ](https://github.com/esphome/esphome/pull/3901)and I'm stuck.  I'm getting output, but it doesn't match Tasmota.  I'm getting small neg ative values (-3.1) instead of positive values.  Other values seem to be close.

emes30, 2023-06-03:
Hi, I fixed negative values. Tasmota's computations can be negative also. But then we've got this:

`if (humidity < 0) humidity = 0.1f;`

so I've just added this to your code
`      if (*humidity < 0)
        *humidity = 0.1f;`

and it works.

ffabrice78, 2023-06-06:
Hi,
Can u send me your yaml, i have the entity but no data.
Thx

emes30, 2023-06-07:
Here is my fork [https://github.com/emes30/esphome/tree/components-dht-ms01](url)
Check it out. If you have entity then your yaml seems ok, check pins and connections.

ffabrice78, 2023-06-10:
Hi,

> external_components:
  - source:
      type: git
      url: https://github.com/emes30/esphome/
      ref: components-dht-ms01
    components: [dht]

switch:
  - platform: gpio
    name: "THR316 Relay"
    pin: GPIO21
    id: relay
  - platform: gpio
    pin: GPIO27
    id: sensor_power
    restore_mode: ALWAYS_ON

sensor:
  - platform: dht
    pin: 25
    model: MS01
    humidity:
     name: "Soil Moisture"
    update_interval: 60s

Here is my yaml, i use ur fork, but i don't have data, just error

emes30, 2023-06-14:
What error ? Can you paste your logs here ?

j-k-olofsson, 2023-06-18:
[10:57:17][C][dht:042]: DHT:
[10:57:17][C][dht:043]:   Pin: GPIO25
[10:57:17][C][dht:049]:   Model: DHT22 (or equivalent)
[10:57:17][C][dht:052]:   Update Interval: 15.0s
[10:57:17][C][dht:054]:   Temperature 'esphome10 Temperature'
[10:57:17][C][dht:054]:     Device Class: 'temperature'
[10:57:17][C][dht:054]:     State Class: 'measurement'
[10:57:17][C][dht:054]:     Unit of Measurement: '°C'
[10:57:17][C][dht:054]:     Accuracy Decimals: 1
[10:57:17][C][dht:055]:   Humidity 'esphome10 Soil Moisture'
[10:57:17][C][dht:055]:     Device Class: 'humidity'
[10:57:17][C][dht:055]:     State Class: 'measurement'
[10:57:17][C][dht:055]:     Unit of Measurement: '%'
[10:57:17][C][dht:055]:     Accuracy Decimals: 0

[10:57:27][W][dht:198]: Requesting data from DHT failed!
[10:57:27][W][dht:085]: Invalid readings! Please check your wiring (pull-up resistor, pin number).

YAML:
sensor:
  - platform: dht
    model: MS01
    pin: GPIO25
    humidity:
      name: ${node_name} Soil Moisture
    temperature:
      name: ${node_name} Temperature
    update_interval: 15s

Coroebus, 2024-05-24:
Hi, 

does SONOFF MS01 is working with ESPHome now ?

Thank You :)

toughvj, 2024-07-19:
hi there, any chance? @emes30  I've tried your solution but it does not work

toughvj, 2024-08-31:
HI @Zebble, maybe U can help me? This is my yaml file:
`substitutions:
  device_name: ogrod-sterownik-nawadniania
  upper_devicename: Ogrod sterownik nawadniania

esphome:
  name: ${device_name}
  friendly_name: ${upper_devicename}
  on_boot:
    - priority: 90
      then:
        - switch.turn_off: relay

esp32:
  board: nodemcu-32s

logger:

api:  
  encryption:    
    key: "***"

ota:
  - platform: esphome
    password: "***"

wifi:
  networks:
    - ssid: !secret wifi_ssid
      password: !secret wifi_psk
      manual_ip:
        static_ip: 10.0.40.230
        gateway: 10.0.40.1
        subnet: 255.255.255.0
      channel: 3
      bssid: "***"
  reboot_timeout: 0s
  fast_connect: True
  power_save_mode: none
  ap:
    ssid: ${upper_devicename}
    password: ""
  on_connect:
    then:
      light.turn_on: wifi_status_led

captive_portal:

web_server:
  port: 80

text_sensor:
  - platform: wifi_info
    ip_address:
      name: Adres IP

external_components:
  - source:
      type: local
      path: ex_components

sensor:
  - platform: dht
    id: dht_sensor
    pin:
      number: GPIO25
      mode:
        input: true
    model: MS01
    temperature: 
[... 167 lines cut ...]
[09:50:53][C][web_server:145]: Web Server:
[09:50:53][C][web_server:146]:   Address: 10.0.40.230:80
[09:50:53][C][mdns:116]: mDNS:
[09:50:53][C][mdns:117]:   Hostname: ogrod-sterownik-nawadniania
[09:50:53][C][esphome.ota:073]: Over-The-Air updates:
[09:50:53][C][esphome.ota:074]:   Address: 10.0.40.230:3232
[09:50:53][C][esphome.ota:075]:   Version: 2
[09:50:53][C][esphome.ota:078]:   Password configured
[09:50:53][C][safe_mode:018]: Safe Mode:
[09:50:53][C][safe_mode:020]:   Boot considered successful after 60 seconds
[09:50:53][C][safe_mode:021]:   Invoke after 10 boot attempts
[09:50:53][C][safe_mode:023]:   Remain in safe mode for 300 seconds
[09:50:53][C][api:139]: API Server:
[09:50:53][C][api:140]:   Address: 10.0.40.230:6053
[09:50:53][C][api:142]:   Using noise encryption: YES
[09:50:53][C][wifi_info:010]: WifiInfo IPAddress 'Adres IP'
[09:50:56][D][api:102]: Accepted 10.0.20.25
[09:50:56][D][api.connection:1391]: Home Assistant 2024.8.3 (10.0.20.25): Connected successfully
[09:51:30][W][dht:198]: Requesting data from DHT failed!
[09:51:30][W][dht:085]: Invalid readings! Please check your wiring (pull-up resistor, pin number).
[09:51:30][D][sensor:094]: 'Temperatura': Sending state nan °C with 1 decimals of accuracy
[09:51:30][D][sensor:094]: 'Wilgotność': Sending state nan % with 0 decimals of accuracy
`

Pins match, the sensor is connected correctly, what else can I do to make it work? The only possibility that comes to my mind is a damaged sensor, my mistake, I didn't check it on the original software, which I unfortunately deleted :( maybe someone has a backup of the firmware for THR316 that they could share?