esphome/feature-requests #1730: Support for Sitronix ST7735S TFT Controller/Driver (Waveshare 0.96" SPI TFT Display)

Scoff123, 2022-05-16:
**Describe the problem you have/What new integration you would like**
I have a Waveshare 0.96" 80x160 TFT SPI Display which uses the ST7735S controller. The screen protector film has a Blue tab. This display is connected to my Lolin D1 Mini Pro ESP8266 board.
Configuring my ESPHome code to use the ST7735 platform doesn't work correctly with my ST7735S Blue tab display.
There is no Blue tab option to specify in the code, so I am using the model INITR_MINI160X80
Using default settings, the resulting display does not fill the screen, and there are borders of undriven screen.
To drive these borders and fill the screen, I have to set the width as 82, and height as 161.

Also, the RGB colours are either inverted or not mapped correctly. For example, if I try to fill the screen with RED only, I define:

color
- id: my_red
  red: 100%
  green: 0%
  blue: 0%

and create a display lambda

lambda: |-
      it.fill(id(my_red));

however the result is a YELLOW screen.

If I create

color
- id: my_white
  red: 100%
  green: 100%
  blue: 100%

the result is a BLACK screen.

If I create 

color
- id: my_black
  red: 0%
  green: 0%
  blue: 0%

the result is a WHITE screen.

**Please describe your use case for this integration and alternatives you've tried:**
I am using this display to show a visual readout of an ultrasonic sensor distance reading. I have erased my ESPHome config from my board and installed Arduino instead, and used the Demo code provided by the manufacturer, and the display works correctly.

If support for this display can be added, It could either be a new supported platform for ST7735S, or alternatively added 
https://esphome.io/components/display/st7735.html

**Additional context**
Details of this display can be found here, including Datasheet and Demo codes for Arduino, Rasp Pi, STM32:

https://www.waveshare.com/product/displays/lcd-oled/lcd-oled-3/0.96inch-lcd-module.htm
https://www.waveshare.com/wiki/0.96inch_LCD_Module
https://www.waveshare.com/wiki/File:0.96inch_LCD_Module_code.7z

yuezhongtao, 2024-04-14:
Is this problem not solved yet in the esphome2023.3.2 version?

nielsnl68, 2024-04-14:
@yuezhongtao do you have still the issues like with above report?

yuezhongtao, 2024-04-16:
> @yuezhongtao do you have still the issues like with above report?

problem has been solved

Joshyakadamien, 2024-09-25:
kinda fixed... on esp32 the display runs smoothly and stable, on 8266 it produces various errors  that seem like an endless rabbit hole. 
If you can move to esp32 and build the project on that it will almost do it out of the box...
the pinout from  this site did the trick properly:
https://www.instructables.com/Value-Your-Project-Use-Graphic-Display/
[image]

Here is the display part of the config which works if on esp32:
(With component IL9XXXX could not get this model to work, 
at best i could get output garbled like when you mis-config the model in the st7735 component.)
```
spi:
  clk_pin: GPIO27   # SCK
  mosi_pin: GPIO14  # SDA

display:
  - platform: st7735
    id: disp
    model: "INITR_BLACKTAB"
    cs_pin: GPIO13  # CS
    dc_pin: GPIO12  # A0
    rotation: 90
    device_width: 128
    device_height: 160
    col_start: 0
    row_start: 0
    eight_bit_color: False
    update_interval: 1s
    spi_mode: MODE3
```