esphome/feature-requests #1731: DSMR P1 Sensor for Watermeter

marcddd, 2022-05-16:
**Describe the problem you have/What new integration you would like**

Is there a possiblity to integrate a sensor for a p1 dsmr watermeter.
**Please describe your use case for this integration and alternatives you've tried:**

I tried to add the Text Sensor "water_equipment_id" but I only get a non comprehensiv long value.
**Additional context**

I use a GWF, modell MTK3coder Mp water meter and has a M-Bus M7. The gas meter runs on channel 1 and the water meter on channel 2 of the m-bus. Both are connected via one cable to a sagecom t-210d. For electricity and gas I get values.

massamino, 2023-06-17:
I'm interested in this as well.
We have a water meter that sends data to the electricity dsmr. The data is there, but needs to go to the dashboard :)

Vato9394, 2023-11-07:
+1, in belgium, they also started to implement this. the data should be available through the P1 port.