esphome/feature-requests #1734: Princess 9000 Smart Air Conditioner (uses ESP32 16MB flash chip)

user897943, 2022-05-17:
**Describe the problem you have/What new integration you would like**

This is a smart aircon enabled by ESP32 chip. I tried the Midea component, doesn't work, maybe it would be compatible with modifications or maybe it requires its own code.

**Please describe your use case for this integration and alternatives you've tried:**

See above
**Additional context**

user897943, 2022-05-17:
> Bad news os is that it looks like the ESP communicates with the other controllers in the system via a secondary UART (U2TXD/U2RXD) so you need the machine in a working condition to test for what commands go there from the ESP towards the rest of the unit. Now that you can't restore original firmware on it, I think it's gonna be a pain...
> 
> Can you at least control it locally with buttons?

I thought so! But not impossible, right?

Yes, there's buttons on the top of the unit and a crappy remote... Really need the smart functions though, that's why I paid more for this unit, just because it had WiFi integrated. I wrote to the manufacturer, hoping they can send a BIN to reflash. I explained the situation.

user897943, 2022-05-17:
Following up from my idea above, I did this:
```
logger:
  baud_rate: 115200

uart:
  tx_pin: 17         # hardware dependant
  rx_pin: 16         # hardware dependant
  baud_rate: 115200
  debug:
    direction: BOTH
    dummy_receiver: true
#    after:
#      delimiter: "\n"
    sequence:
      - lambda: UARTDebug::log_string(direction, bytes);
```

and got this after pressing different buttons on the remote, looks like there is a length pattern... although it's all zeros!? I'm no coder but I can see that each button press has a respective quantity of zeros
```
INFO Reading configuration /config/airconditioner-fb1.yaml...
INFO Detected timezone 'Etc/UTC'
INFO Generating C++ source...
INFO Compiling app...
Processing airconditioner-fb1 (board: widora-air; framework: arduino; platform: platformio/espressif32 @ 3.5.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 16MB Flash
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
Dependency Graph
|-- <WiFi> 1.0
|-- <ESPmDNS> 1.0
|-- <Update> 1.0
Compiling .pioenvs/airconditioner-fb1/src/main.cpp.o
Linking .pioenvs/airconditioner-fb1/firmware.elf
RAM:   [=         ]  12.6% (used 41184 bytes from 327680 bytes)
Flash: [=====     ]  45.5% (used 835150 bytes from 1835008 bytes)
Building .pioenvs/airconditioner-fb1/firmware.bin
esp32_create_combined_bin([".pioenvs/airconditioner-fb1/firmware.bin"], [".pioenvs/airconditioner-fb1/firmware.elf"])
Flash params set to 0x0240
Wrote 0xdbec0 bytes to file /config/.esphome/build/airconditioner-fb1/.pioenvs/airconditioner-fb1/firmware-factory.bin, ready to flash to offset 0x0
========================= [SUCCESS] Took 14.66 seconds =========================
INFO Successfully compiled program.
INFO Connecting to 10.0.0.151
INFO Uploading /config/.esphome/build/airconditioner-fb1/.pioenvs/airconditioner-fb1/firmware.bin (835264 bytes)
Uploading: [============================================================] 100% Done...

INFO Waiting for result...
INFO OTA successful
INFO Successfully uploaded program.
INFO Starting log output from 10.0.0.151 using esphome API
INFO Successfully connected to 10.0.0.151
[17:25:15][I][app:102]: ESPHome version 2022.6.0-dev compiled on May 17 2022, 17:24:46
[17:25:15][C][wifi:491]: WiFi:
[17:25:15][C][wifi:353]:   Local MAC: A4:E5:7C:52:89:A0
[17:25:15][C][wifi:354]:   SSID: [redacted]
[17:25:15][C][wifi:355]:   IP Address: 10.0.0.151
[17:25:15][C][wifi:357]:   BSSID: [redacted]
[17:25:15][C][wifi:358]:   Hostname: 'airconditioner-fb1'
[17:25:15][C][wifi:360]:   Signal strength: -47 dB ▂▄▆█
[17:25:15][C][wifi:364]:   Channel: 11
[17:25:15][C][wifi:365]:   Subnet: 255.255.255.0
[17:25:15][C][wifi:366]:   Gateway: 10.0.0.1
[17:25:15][C][wifi:367]:   DNS1: 10.0.0.1
[17:25:15][C][wifi:368]:   DNS2: 10.0.0.120
[17:25:15][C][logger:274]: Logger:
[17:25:15][C][logger:275]:   Level: DEBUG
[17:25:15][C][logger:276]:   Log Baud Rate: 115200
[17:25:15][C][logger:277]:   Hardware UART: UART0
[17:25:15][C][uart.arduino_esp32:107]: UART Bus:
[17:25:15][C][uart.arduino_esp32:108]:   TX Pin: GPIO17
[... 55 lines cut ...]
[17:25:33][C][api:143]:   Using noise encryption: NO
[17:25:33][C][sntp:050]: SNTP Time:
[17:25:33][C][sntp:051]:   Server 1: '0.pool.ntp.org'
[17:25:33][C][sntp:052]:   Server 2: '1.pool.ntp.org'
[17:25:33][C][sntp:053]:   Server 3: '2.pool.ntp.org'
[17:25:33][C][sntp:054]:   Timezone: 'UTC0'
[17:25:51][D][uart_debug:158]: <<< "\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00"
[17:25:55][D][uart_debug:158]: <<< "\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00"
[17:26:00][D][uart_debug:158]: <<< "\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00"
[17:26:05][D][uart_debug:158]: <<< "\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00"
[17:26:13][D][uart_debug:158]: <<< "\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00"
[17:26:23][D][uart_debug:158]: <<< "\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00"
[17:26:25][D][uart_debug:158]: <<< "\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00"
[17:26:30][D][uart_debug:158]: <<< "\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00"
[17:26:32][D][uart_debug:158]: <<< "\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00"
[17:26:36][D][uart_debug:158]: <<< "\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00"
[... cut ...] x00"
[17:26:58][D][uart_debug:158]: <<< "\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00"
[17:27:07][D][uart_debug:158]: <<< "\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00"
```

Any ideas? I think this might help finding the control codes!?

nagyrobi, 2022-05-17:
Nice results, now try sending back the same codes, if you're lucky enough it may work. Otherwise what you see are just status updates from the main MCU intended towards the WiFi module... See: 
https://esphome.io/components/switch/uart.html
https://esphome.io/components/uart.html#uart-write-action
https://esphome.io/custom/uart.html
https://esphome.io/cookbook/uart_text_sensor.html

user897943, 2022-05-17:
> Nice results, now try sending back the same codes, if you're lucky enough it may work. Otherwise what you see are just status updates from the main MCU intended towards the WiFi module... See: [...]

Don't know how to send it, or even if the codes are formatted correctly... I have a medical procedure tomorrow so can't review the stuff and do more today but will be reading the info at a later date. Would like to keep an eye on this and see if someone can jump in and perhaps adapt the MIDEA component? It's probably very similar just using different codes?

user897943, 2022-05-21:
Back from the hospital, upon some googling it looks like the PCB inside the unit is some sort of MIDEA kind, further googling revealed this, was wondering if someone with coding experience could compile something and give to me to flash to the ESP32 and test? I will be absolutely happy to test the code! I just don't know how to write it. I am doing everything I possibly can like researching. https://github.com/uncle-yura/esphome-midea-ac

kquinsland, 2022-06-06:
A few things come to mind:

```
[10:18:38]I (9730) mcu_write: Sending 7 bytes (version=0, command=0, datalen=0):
[10:18:38]I (9730) mcu_write: 0x3ffbb598   55 aa 00 00 00 00 ff                              |U......|
[10:18:43]I (14860) MQTT_CLIENT: Sending MQTT CONNECT message, type: 1, id: 0000
[10:18:43]I (14920) cloud: Subscribing: appliance/airconditioner/3c39e7248f22/$update/set
```

the `55aa...` preamble has triggered some 'pattern matching' brain cells. This is the same preamble that TuYa uses.

And, that seems to check out / match.

I did a quick 'spot check' with some of the data you supplied:

```

https://developer.tuya.com/en/docs/iot/mcu-protocol?id=K9hrdpyujeotg

turn off:           55:AA:00:07:00:05:01:01:00:01:00:0E
turn on:            55:AA:00:07:00:05:01:01:00:01:01:0F

fan speed low:      55:AA:00:07:00:05:05:04:00:01:00:15
fan speed medium:   55:AA:00:07:00:05:05:04:00:01:01:16
fan speed high:     55:AA:00:07:00:05:05:04:00:01:02:17
```

The last 2 bytes of each (`0e`, `0f`, `15`,`16`,`17`) all match a basic module256 checksum as outlined in the TuYa docs (you can use any online calc to quickly check this. e.g.: https://www.scadacore.com/tools/programming-calculators/online-checksum-calculator/)

So the good news is that ESPHome / Tasmota communities have already done EXTENSIVE RE work on TuYa. There is an ESPHome TuYa component that should implement _most_ of the boilreplate code.

Read the tuya docs on the packet format and apply the learning (e.g.: the data point / params) to the paylaods you've pulled out already. Spreadsheets can be a useful tool for hilighting changes between packets. ESPHome has the code to generate the header and other crap for the packets you'll need to send to the MCU... but you'll have to figure out the details: which datapoint(s) control the ... fan, compressor ... etc.

----

As for firmware restoration:

First, do a _full_ backup of the _entire_ chip. Do this 3 times, compare the file hashes of each and make sure they match.
Replace the first N bytes of this dump with the partial dump you took initially.  Flash this back to the unit and give it a shot.

This is a hailmary but - if ESPHome didn't modify anything in flash beyond the 4MB boundary - it should work.

Alternatively, eBay or similar eWaste places might be a way to get a used/broken unit from which you can get a 'known good' ESP module.

kquinsland, 2022-06-06:
And did a bit more diffing:

https://developer.tuya.com/en/docs/iot/mcu-protocol?id=K9hrdpyujeotg

```
fan speed low:      55:AA:                          # HEADER
                    00:                             # VERSION
                    07:                             # COMMAND
                    00:05:                          # LEN
                    05:04:00:01:00:                 # DATA
                    15                              # CHKSM

The important bits:
                    05:04:00:01:00                  # DATA, LOW
                    05:04:00:01:01                  # DATA, MEDIUM
                    05:04:00:01:02                  # DATA, HIGH

```

Compare with one of the examples from other repo:

```
  fan_mode_low_action:
  - homeassistant.service:
      service:  esphome.esp_conditioner_state
      data: 
        speed: "10"     # Or 25, 100
        mode: "-1"
        power: "-1"
        vswing: "-1"
        hswing: "-1"
        temp: "-1"
```

The data: is passed to this code:

https://github.com/uncle-yura/esphome-midea-ac/blob/main/midea.cpp#L218

which does not *at least with a super quick look* implement TuYa protocol / it's not clear how 

- speed: "10"
- speed: "25"
- speed: "100"

would map back to

- 05:04:00:01:00
- 05:04:00:01:01
- 05:04:00:01:02

So I'm not sure that the other repo you've found is implementing the same protocol... but I am also doing this quickly while I watch the WWDC keynote in the background so my super quick read might be wrong.

user897943, 2022-06-06:
> > Edit: tried my best, could not do it (sending codes) but I tested all the buttons from the panel and remote and recorded everything below. Would be awesome if someone would help fill the gaps.
> 
> This is how you would send bytes back down: https://github.com/kquinsland/lg-m43mu79-esp-home-bridge/blob/master/esphome/monitor-bridge.yaml#L198
> 
> Try with a simple template switch so you can access it via the web UI:
> 
> ```yaml
> switch:
>   - platform: template
>     name: "testing"
>     turn_on_action:
>       then:
>         - uart.write: [0x55, 0xAA, 0x00, 0x07, 0x00, 0x05, 0x01, 0x01, 0x00, 0x01, 0x01, 0x0F]
>     turn_off_action:
>       then:
>         - uart.write: [0x55, 0xAA, 0x00, 0x07, 0x00, 0x05, 0x01, 0x01, 0x00, 0x01, 0x00, 0x0E]
> ```
> 
> Try repeatedly; if anything else is "talking" when you're trying to transmit, the receiver(s) will get confused and likely just ignore the garbled message they get. Likewise, there _might_ be some "initalization" work you need to do (from the logs you shared:)
> 
> ```
> Showing logs:
> [10:18:30]I (2260) tcpip_adapter: sta ip: 10.0.0.159, mask: 255.255.255.0, gw: 10.0.0.1
> [10:18:30]I (2260) mcu_write: Sending 8 bytes (version=0, command=3, datalen=1):
> [10:18:30]I (2260) mcu_write: 0x3ffb9fa4   55 aa 00 03 00 01 03 06                           |U.......|
> [10:18:38]I (9730) mcu_write: Sending 7 bytes (version=0, command=0, datalen=0):
> [10:18:38]I (9730) mcu_write: 0x3ffbb598   55 aa 00 00 00 00 ff      
> ```
> 
> You might use `on_boot:` trigger from esphome to `uart.write: [0x55, 0xaa, 0x00, 0x00, 0x00, 0x00, 0xff ]` or similar if the template switch above doesn't work.

I did this on my yaml:
```
switch:
  - platform: template
    name: "testing"
    turn_on_action:
      then:
        - uart.write:
            id: uart_1
            data: [0x55, 0xAA, 0x00, 0x07, 0x00, 0x05, 0x01, 0x01, 0x00, 0x01, 0x01, 0x0F]
    turn_off_action:
      then:
        - uart.write:
            id: uart_1
            data: [0x55, 0xAA, 0x00, 0x07, 0x00, 0x05, 0x01, 0x01, 0x00, 0x01, 0x00, 0x0E]
```

and this is what I get on the logs (switch acts like a momentary, doesn't stay on, always goes back to off, there's no change on the AC, and I tried uart1 and uart2.

```
9:34:28][C][ota:086]:   Address: 10.0.0.151:3232
[19:34:28][C][ota:089]:   Using Password.
[19:34:28][C][api:138]: API Server:
[19:34:28][C][api:139]:   Address: 10.0.0.151:6053
[19:34:28][C][api:143]:   Using noise encryption: NO
[19:34:30][D][uart_debug:114]: <<< 00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
[19:34:31][D][api:102]: Accepted ::FFFF:A00:D3
[19:34:31][D][api.connection:861]: Home Assistant 2022.6.2 (::FFFF:A00:D3): Connected successfully
[19:34:33][D][uart_debug:114]: <<< 00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
[19:34:36][D][switch:013]: 'testing' Turning ON.
[19:34:36][D][uart_debug:114]: <<< 00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
[19:34:36][D][uart_debug:114]: >>> 55:AA:00:07:00:05:01:01:00:01:01:0F
[19:34:39][D][uart_debug:114]: <<< 00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
[19:34:39][D][switch:013]: 'testing' Turning ON.
[19:34:39][D][uart_debug:114]: <<< 00:00:00
[19:34:39][D][uart_debug:114]: >>> 55:AA:00:07:00:05:01:01:00:01:01:0F
[19:34:42][D][uart_debug:114]: <<< 00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
[... cut ...] :00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
[19:35:06][D][uart_debug:114]: <<< 00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
```
For speed's sake, how should I try the on_boot thing? How should I do the initialization? Thanks!

user897943, 2022-06-06:
I did this on the yaml:
```
esphome:
  name: airconditioner-fb1
  platform: ESP32
  board: widora-air
  on_boot:
    priority: 600
    then:
      - uart.write: 
          id: uart_1
          data: [ 0x55, 0xaa, 0x00, 0x00, 0x00, 0x00, 0xff ]
```

getting this on the logs (still no luck, tried uart1 and 2):
```
[19:42:07][C][mdns:085]:   Hostname: airconditioner-fb1
[19:42:07][C][ota:085]: Over-The-Air Updates:
[19:42:07][C][ota:086]:   Address: 10.0.0.151:3232
[19:42:07][C][ota:089]:   Using Password.
[19:42:07][C][api:138]: API Server:
[19:42:07][C][api:139]:   Address: 10.0.0.151:6053
[19:42:07][C][api:143]:   Using noise encryption: NO
[19:42:09][D][uart_debug:114]: <<< 00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
[19:42:10][D][api:102]: Accepted ::FFFF:A00:D3
[19:42:10][D][api.connection:861]: Home Assistant 2022.6.2 (::FFFF:A00:D3): Connected successfully
[19:42:12][D][uart_debug:114]: <<< 00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
[19:42:15][D][uart_debug:114]: <<< 00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
[19:42:15][D][switch:013]: 'testing' Turning ON.
[19:42:15][D][uart_debug:114]: <<< 00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
[19:42:15][D][uart_debug:114]: >>> 55:AA:00:07:00:05:01:01:00:01:01:0F
[19:42:18][D][uart_debug:114]: <<< 00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
[19:42:21][D][uart_debug:114]: <<< 00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
```

user897943, 2022-06-06:
I googled the hex code and found this:
https://images.tuyacn.com/smart/aircondition/Guide-to-Interworking-with-the-Tuya-MCU.pdf

But absolutely zero idea on how to work with that. It does seem to make sense for the AC though...

kquinsland, 2022-06-06:
> getting this on the logs (still no luck, tried uart1 and 2):

Reading in all 0s is suspect. Use the same UART config you used when you were able to record this: https://github.com/esphome/feature-requests/issues/1734#issuecomment-1134483630

(more on this in a sec w/ `2` below)

---

Since it's likely the rest of flash is gone, I see three options:

1. Get another unit, dump flash. The first post you had in this thread made it look like the OEM firmware uses MQTT. If they're not doing things securely, you might be able to trick the OEM firmware into communicating with your own MQTT broker. You will almost certainly want to dump the traffic first to understand what - for example - needs to be written to the `appliance/airconditioner/3c39e7248f22/state/power_on/set` topic. If they did do things properly (read: TLS and cert pinning) then this will be _much_ more difficult.
2. You recorded those payloads from the UART as you were pressing buttons on the remote, right? Looking under the `Command index` section from here: https://developer.tuya.com/en/docs/iot/mcu-protocol?id=K9hrdpyujeotg, packet type `0x07` is the "you should know" type which makes sense: the MCU is telling the ESP that the state has changed.  Commanding the MCU requires packets of type `0x06` which is `Send the DP commands` and `0x07` is `Report the DP status`. Unfortunately, it's still a bit of a brute-force / guess and check approach to figuring out what datapoints IDs exist and what format they want data in. Tasmota docs have some useful pointers about how to do this: https://tasmota.github.io/docs/TuyaMCU/#fnid.  There might be some overlap with the 'known/common' datapoints and function IDs here: https://tasmota.github.io/docs/TuyaMCU/#air-purifier
3. Implement an IR blaster. ESPHome has support for this and there are a billion guides out there that will go over how to hook up an IR transmitter and receiver to an ESP module so you can record/play-back the remote commands. This has all the drawbacks of an "open loop" system so it's not ideal... but it's certainly the simplest to do from a technical perspective.

You could also just try this:

```yaml
        - uart.write:
            id: uart_1
            # 55:AA:00:06:00:05:01:01:00:01:01
            data: [0x55, 0xAA, 0x00, 0x06, 0x00, 0x05, 0x01, 0x01, 0x00, 0x01, 0x01, 0x0E]
    turn_off_action:
      then:
        - uart.write:
            id: uart_1
            # 55:AA:00:06:00:05:01:01:00:01:00
            data: [0x55, 0xAA, 0x00, 0x06, 0x00, 0x05, 0x01, 0x01, 0x00, 0x01, 0x00, 0x0D]
```

All I did was change the `0x07` to `0x06` and re-compute the checksum. This is essentially what @nagyrobi  suggested here: https://github.com/esphome/feature-requests/issues/1734#issuecomment-1129187713

---

Its been a LONG time since i did any TuYa RE work, but it looks like the init stuff:

```
[10:18:38]I (9730) mcu_write: Sending 7 bytes (version=0, command=0, datalen=0):
[10:18:38]I (9730) mcu_write: 0x3ffbb598   55 aa 00 00 00 00 ff      
```
Can be explained with the `Detect the heartbeat` section from the same (above) link.

```
For example,

    The module sends: 55 aa 00 00 00 00 ff
    The MCU returns:
        The first time: 55 aa 03 00 00 01 00 03
        The second time and afterward: 55 aa 03 00 00 01 01 04
```

This has me 100% convinced that internally, this thing is meant for use with TuYa wifi chips!

Just below that section is this:

```
For example,

    The module sends: 55 aa 00 01 00 00 00
    The MCU returns: 55 aa 03 01 00 2a 7b 22 70 22 3a 22 52 4e 32 46 56 41 67 58 47 36 57 66 41 6b 74 55 22 2c 22 76 22 3a 22 31 2e 30 2e 30 22 2c 22 6d 22 3a 30 7d 0c

        Note: Convert each character (including punctuation marks) in {"p":"RN2FVAgXG6W****","v":"1.0.0","m"} into ASCII code, and enter it in the data position.

```

So I would do that.... Send those bytes (55 aa 00 01 00 00 00) down the UART and see what comes back. Decode the bytes into ASCII text and whatever value for `p` comes back might be something that you can google.  Every TuYa device model has a unique value and once you figure out what yours is... google might show you somebody that has already done the work.  More than likely, this appliance is made by somebody else and `princess` just licensed it / stuck their name on the box / implemented their own firmware for the ESP32 but kept all the internal uart protocols intact.  This would explain why searching for existing work on the `princess 9000` model returns nothing. If we knew what model this thing _really_ is then there might be some better results.

user897943, 2022-06-06:
OMG. IT WORKED! 
Bits of the yaml:
```
switch:
  - platform: template
    name: "testing"
    turn_on_action:
      then:
        - uart.write:
            id: uart_2
            data: [0x55, 0xAA, 0x00, 0x06, 0x00, 0x05, 0x01, 0x01, 0x00, 0x01, 0x01, 0x0E]
    turn_off_action:
      then:
        - uart.write:
            id: uart_2
            data: [0x55, 0xAA, 0x00, 0x06, 0x00, 0x05, 0x01, 0x01, 0x00, 0x01, 0x00, 0x0D]
```
and
```
  on_boot:
    priority: 600
    then:
      - uart.write: 
          id: uart_2
          data: [ 0x55, 0xaa, 0x00, 0x00, 0x00, 0x00, 0xff ]
```
The AC turned on and off when I used the switch in HA. I am begging you to code this for me and other users of this AC. I will do any necessary testing to assist you as best as I can. I just can't understand all the technical stuff involved in a reasonable timeframe and I personally need this AC controllable now more than ever with my recovery...

kquinsland, 2022-06-07:
> OMG. IT WORKED!

Awesome.

Try with out the `on_boot`. I suspect that it'll still work.... but if not, that's good to learn.

- "it worked" meaning that the **entire** unit went on/off as if you had hit the power button? or just the AC portion went off (but the fan kept running...?)

- What is "sleep mode"?

- What is `swing` ?  Is this the oscillation? 
---

Enable/configure the tuya component: https://esphome.io/components/tuya.html and dump **all** the logs and share here.

Get set up the same way you were when you did this: https://github.com/esphome/feature-requests/issues/1734#issuecomment-1134483630

Once I found this: https://tasmota.github.io/docs/TuyaMCU/#data-type-table some of the packets were easy/obvious to decode but there are some things that are still confusing. See if you can figure out where the temperature sensor is because you need to get some temperature reading device in the same vicinity.  I am pretty sure you found how to tell the machine what temperature to get to, but we need to identify what packets the machine reports (if any) that state what temperature it thinks it is. 

If I had to guess, it'll be whatever is plugged in to the `P_T-EVA` socket. I suspect it'll look similar to whatever is plugged into the `P_T-CON` socket but that will almost certainly be attached to the little radiator that has all the heat / that gets blown outside.  Try to find this temp sensor and touch it / hold it in your hand / blow on it / use a hair dryer or something to artificially get it it supper warm... Hopefully that'll result in some log messages getting spewed out which might be enough to complete the puzzle:

```yaml
climate:
  - platform: tuya
    name: "My Climate Device"
    switch_datapoint: 1
    supports_heat: false
    supports_cool: true
    

    # packet, 16c: 04:02:00:04:00:00:00:10
    # packet, 17c: 04:02:00:04:00:00:00:11
    #
    # DP: 4, type 2 (4 byte data) and 0x10 is 16, 0x11 is 17 so it's
    #   safe to say that the data is just the desired temp in c.
    ##
    target_temperature_datapoint: 4
    
    # Still need to RE this
    current_temperature_datapoint: 
```

But for right now, just the fan bits:

```yaml
fan:
  - platform: "tuya"
    name: "MyFan"
    switch_datapoint: 1
    ##
    # L   05:04:00:01:00
    # M   05:04:00:01:01
    # H   05:04:00:01:02
    # DP: 5
    # TYPE: ENUM
    ##
    speed_datapoint: 5
    
    # Not sure. Might have to just do power which is DP1
    ##
    # turn off:   01:01:00:01:00
    # turn on:    01:01:00:01:01
    # DP: 1
    # type 1
    ##
    switch_datapoint: 1
    
    # Is this swing mode. If NO, comment out!
    ##
    # Swing on:   06:01:00:01:01
    # Swing off:  06:01:00:01:00
    # DP: 6
    # type 1
    ##
    oscillation_datapoint: 6
```

If at all possible, get the machine into fan only mode and then try to turn the fan on/off WITHOUT powering the whole thing down... if possible. If the only way to stop the fan in "fan only" mode is to turn the whole thing off, then keep the `switch_datapoint: 1`

user897943, 2022-06-07:
> Get set up the same way you were when you did this: #1734 (comment)

Thanks, I am currently in hospital due to a hemorrhage so I won't be able
to follow the instructions until I get back home. Thanks for this, I hope
to do it soon.

user897943, 2022-06-08:
Back home, and some good news...
[image]

When I toggled the switch, the AC went on/off on the last selected mode. It doesn't change the mode at all. Acts as a on/off only. Swing is the flap on top, if it's on the flap oscillates, if it's off the flap stays fully open (or whichever position it was when you set swing/oscillation to off). There's only vertical oscillation.

Sleep mode is like a silent mode... I found this after googling:
"This function is useful for night as it gradually reduces the operation of the air conditioner 
when running in cooling mode. The air conditioner will run with a low fan speed and the 
desired temperature will increase by 1°C per hour for two hours ( i.e max of 2°C above 
temperature originally set). The unit will then continue to run set at this temperature for a 
further 6 hours before entering standby mode."

The remote does not show current temp.

There's definitely a temp sensor on the back coils, and another on the output I think. I will blow some hot air while the log is open and see what happens.

This is my yaml:
```
esphome:
  name: airconditioner-fb1
  platform: ESP32
  board: widora-air
#  on_boot:
#    priority: 600
#    then:
#      - uart.write: 
#          id: uart_2
#          data: [ 0x55, 0xaa, 0x00, 0x00, 0x00, 0x00, 0xff ]

wifi:
  ssid: "xxx"
  password: "xxx"
  manual_ip:
    static_ip: 10.0.0.151
    gateway: 10.0.0.1
    subnet: 255.255.255.0
    dns1: 10.0.0.1
    dns2: 10.0.0.120  

# Enable Home Assistant API
api:
  password: !secret passapi
  
ota:
  password: !secret passota

# Enable logging
logger:
  level: DEBUG
  baud_rate: 0

uart:
  - id: uart_1
    tx_pin: 1 #was 1
    rx_pin: 3 #was 3
    baud_rate: 9600
    debug:
      direction: BOTH
      dummy_receiver: true
      after:
        delimiter: "\n"
  - id: uart_2
    tx_pin: 17
    rx_pin: 16
    baud_rate: 9600
    debug:
      direction: BOTH
      dummy_receiver: true
      after:
        delimiter: "\n"
[... 109 lines cut ...]
[18:27:44][C][tuya:039]:   Configuration will be reported when setup is complete. Current init_state: 0
[18:27:44][C][tuya:041]:   If no further output is received, confirm that this is a supported Tuya device.
[18:27:44][D][uart_debug:114]: <<< 55:AA:00:00:00:01:01:01
[18:27:44][D][uart_debug:114]: >>> 55:AA:00:00:00:00:FF
[18:27:44][D][uart_debug:114]: <<< 55:AA:00:00:00:01:01:01
[18:27:44][E][tuya:409]: Initialization failed at init_state 0
[18:27:45][D][uart_debug:114]: <<< 00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
[18:27:46][D][api:102]: Accepted ::FFFF:A00:D3
[18:27:46][D][api.connection:861]: Home Assistant 2022.6.4 (::FFFF:A00:D3): Connected successfully
[18:27:48][D][uart_debug:114]: <<< 00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
[18:27:51][D][uart_debug:114]: <<< 00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
[18:27:53][D][uart_debug:114]: >>> 55:AA:00:00:00:00:FF
[18:27:53][D][uart_debug:114]: <<< 55:AA:00:00:00:01:01:01
[18:27:53][D][uart_debug:114]: >>> 55:AA:00:00:00:00:FF
[18:27:54][D][uart_debug:114]: <<< 55:AA:00:00:00:01:01:01
[18:27:54][D][uart_debug:114]: >>> 55:AA:00:00:00:00:FF
[18:27:54][D][uart_debug:114]: <<< 55:AA:00:00:00:01:01:01
[18:27:54][D][uart_debug:114]: >>> 55:AA:00:00:00:00:FF
[18:27:54][D][uart_debug:114]: <<< 00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
[18:27:54][D][uart_debug:114]: <<< 55:AA:00:00:00:01:01:01
[18:27:54][D][uart_debug:114]: >>> 55:AA:00:00:00:00:FF
[18:27:55][D][uart_debug:114]: <<< 55:AA:00:00:00:01:01:01
[18:27:55][E][tuya:409]: Initialization failed at init_state 0
```
Toggling the fan switch only causes the unit to emit several beeps in succession, it doesn't do anything. So I believe this is definitely based on Tuya and we're on the right track.

kquinsland, 2022-06-08:
> [18:27:55][E][tuya:409]: Initialization failed at init_state 0

That's not good. I was expecting something like this in the logs:

```
[12:39:45][C][tuya:023]: Tuya:
[12:39:45][C][tuya:032]:   Datapoint 1: switch (value: ON)
[12:39:45][C][tuya:036]:   Datapoint 3: enum (value: 1)
[12:39:45][C][tuya:036]:   Datapoint 6: enum (value: 0)
[12:39:45][C][tuya:034]:   Datapoint 7: int value (value: 0)
[12:39:45][C][tuya:032]:   Datapoint 9: switch (value: OFF)
[12:39:45][C][tuya:046]:   Product: '{"p":"hqq73kftvzh8c92u","v":"1.0.0","m":0}'
```

Specifically wanted the `"p":"..."` value.  Possibly disable the uart debugger? It might not be fully tuya compliant, too :/.

> When I toggled the switch, the AC went on/off on the last selected mode. It doesn't change the mode at all. Acts as a on/off only.

That makes sense. It would be odd if the compressor was still on but the fan was off. This datapoint is "whole unit power" then.
To do anything fancy (like run the fan for a few min after the compressor has turned off to get the most cool air possible from the evaporator) you'll have to manually switch modes then count down for a few min before turning the unit off all together.

> There's definitely a temp sensor on the back coils, and another on the output I think. I will blow some hot air while the log is open and see what happens.

I think the `EVA` and `CND` are evaporator and condenser. Find whatever is plugged in to the `IN/T` socket: https://user-images.githubusercontent.com/5891221/168818338-95a441c4-f2ae-45ef-b06e-6fb6dccae3cc.jpg

It might also just be on the motherboard.

That's the interior temp senor that you need to blast with hot air. You should see the temp display on the unit also rise to match. Once it does, start looking at the logs.

It'll be some packet starting with `>>> 55:AA:00:07:` and like when you spotted the "increases by 1" pattern with the temperature there should be some octet that changes as the temp does. higher numbers when the hot air is on, falling numbers when the air is off. repeat a few times to confirm.  

Find the packets that flow when the temp sensor notices a new room temp ... if there are any. If you could see this with the "stock" app then there for sure are. If the stock app never did show the current room temp... that's a bad sign.

After finding the packets, figure out how the temp is encoded. Then you should have a functional fan/climate entity.

reverse engineering sleep mode and all the other crap is an exercise left to the reader. With functional primitives, it's easier to re-implement in ESPhome

user897943, 2022-06-08:
> That's not good. I was expecting something like this in the logs:
>
> ```
> [12:39:45][C][tuya:023]: Tuya:
> [12:39:45][C][tuya:032]:   Datapoint 1: switch (value: ON)
> [12:39:45][C][tuya:036]:   Datapoint [...]

I just noticed this and still have to read it, in the meantime please see this, could this help in any way?: https://developer.tuya.com/en/docs/iot/product-function-definition?id=K9tp14pijmeyh

user897943, 2022-06-08:
Ok, from the link above I modified the yaml.
```
uart:
  - id: uart_1
    tx_pin: 1 #was 1
    rx_pin: 3 #was 3
    baud_rate: 9600
#    debug:
#      direction: BOTH
#      dummy_receiver: true
#      after:
#        delimiter: "\n"
  - id: uart_2
    tx_pin: 17
    rx_pin: 16
    baud_rate: 9600
#    debug:
#      direction: BOTH
#      dummy_receiver: true
#      after:
#        delimiter: "\n"

switch:
  - platform: template
    name: "testing"
    turn_on_action:
      then:
        - uart.write:
            id: uart_2
            data: [0x55, 0xAA, 0x00, 0x06, 0x00, 0x05, 0x01, 0x01, 0x00, 0x01, 0x01, 0x0E]
    turn_off_action:
      then:
        - uart.write:
            id: uart_2
            data: [0x55, 0xAA, 0x00, 0x06, 0x00, 0x05, 0x01, 0x01, 0x00, 0x01, 0x00, 0x0D]

tuya:
  uart_id: uart_2

climate:
  - platform: tuya
    name: "My Climate Device"
    switch_datapoint: 1
    supports_heat: false
    supports_cool: true
    

    # packet, 16c: 04:02:00:04:00:00:00:10
    # packet, 17c: 04:02:00:04:00:00:00:11
    #
    # DP: 4, type 2 (4 byte data) and 0x10 is 16, 0x11 is 17 so it's
    #   safe to say that the data is just the desired temp in c.
    ##
    target_temperature_datapoint: 16
    
    # Still need to RE this
    #current_temperature_datapoint:       

fan:
  - platform: "tuya"
    name: "MyFan"
    ##
    # L   05:04:00:01:00
    # M   05:04:00:01:01
    # H   05:04:00:01:02
    # DP: 5
    # TYPE: ENUM
    ##
    speed_datapoint: 17
    
[... 86 lines cut ...]
[19:59:16][C][tuya:054]:   Datapoint 5: enum (value: 2)
[19:59:16][C][tuya:048]:   Datapoint 6: switch (value: ON)
[19:59:16][C][tuya:050]:   Datapoint 7: int value (value: 0)
[19:59:16][C][tuya:056]:   Datapoint 8: bitmask (value: 0)
[19:59:16][C][tuya:048]:   Datapoint 9: switch (value: OFF)
[19:59:16][C][tuya:048]:   Datapoint 10: switch (value: OFF)
[19:59:16][C][tuya:056]:   Datapoint 50: bitmask (value: 0)
[19:59:16][C][tuya:068]:   Product: ''
[19:59:16][C][tuya:033]: Tuya:
[19:59:16][C][tuya:048]:   Datapoint 1: switch (value: OFF)
[19:59:16][C][tuya:054]:   Datapoint 2: enum (value: 0)
[19:59:16][C][tuya:050]:   Datapoint 3: int value (value: 20)
[19:59:16][C][tuya:050]:   Datapoint 4: int value (value: 16)
[19:59:16][C][tuya:054]:   Datapoint 5: enum (value: 2)
[19:59:16][C][tuya:048]:   Datapoint 6: switch (value: ON)
[19:59:16][C][tuya:050]:   Datapoint 7: int value (value: 0)
[19:59:16][C][tuya:056]:   Datapoint 8: bitmask (value: 0)
[19:59:16][C][tuya:048]:   Datapoint 9: switch (value: OFF)
[19:59:16][C][tuya:048]:   Datapoint 10: switch (value: OFF)
[19:59:16][C][tuya:056]:   Datapoint 50: bitmask (value: 0)
[19:59:16][C][tuya:068]:   Product: ''
[19:59:19][D][api:102]: Accepted ::FFFF:A00:D3
[19:59:19][D][api.connection:861]: Home Assistant 2022.6.4 (::FFFF:A00:D3): Connected successfully
[19:59:59][D][esp32.preferences:114]: Saving preferences to flash...**
```

user897943, 2022-06-08:
> Ok, from the link above I modified the yaml.
>
> ```
> uart:
>   - id: uart_1
>     tx_pin: 1 #was 1
>     rx_pin: 3 #was 3
>     baud_rate: 9600
> #    debug:
> #      direction: BOTH
> #      dummy_receiver: true [...]

Just tested the above. Seems that using the datapoints as per the link is kinda working! Now the fan entity oscillate button works. I can toggle swing on and off, and the LED on the AC even goes on or off accordingly, and the flap oscillates or stops as expected.
The fan speed can't be changed on the fan entity.
The AC entity has a cool mode and off mode. The off mode turns the whole thing off. The cool mode doesn't do anything. The I am now getting set temp and current temp but they do not change, it's always 32 degrees Fahrenheit.

So we are getting somewhere and this is absolutely Tuya. I can't modify code, unfortunately. If nobody with programming knowledge is willing to help me, then this will be a great loss I think, because I'm definitely trying my hardest here, I only need help from more experienced persons to fill the gaps.

user897943, 2022-06-08:
I have mostly good news. It's mostly working!

Here's the yaml and what I'm getting:
```
esphome:
  name: airconditioner-fb1
  platform: ESP32
  board: widora-air

wifi:
  ssid: "Kumshott"
  password: "isubmittopeter"
  manual_ip:
    static_ip: 10.0.0.151
    gateway: 10.0.0.1
    subnet: 255.255.255.0
    dns1: 10.0.0.1
    dns2: 10.0.0.120  

# Enable Home Assistant API
api:
  password: !secret passapi
  
ota:
  password: !secret passota

# Enable logging
logger:
  level: DEBUG
  baud_rate: 0

uart:
  - id: uart_1
    tx_pin: 1
    rx_pin: 3
    baud_rate: 9600

  - id: uart_2
    tx_pin: 17
    rx_pin: 16
    baud_rate: 9600

tuya:
  uart_id: uart_2

climate:
  - platform: tuya
    name: "Air conditioner - Front room"
    switch_datapoint: 1
    supports_heat: false
    supports_cool: true
    target_temperature_datapoint: 3
    current_temperature_datapoint: 4
    eco_datapoint: 7
    active_state_datapoint: 2
    active_state_cooling_value: 0

fan:
  - platform: "tuya"
    name: "Air conditioner (Fan) - Front room"
    speed_datapoint: 5
    switch_datapoint: 1
    oscillation_datapoint: 6
```
These are working: turning on/off from the climate component. Turning on/off from the fan component. Setting fan speed. Setting oscillation. There's a datapoint 9 (I believe) for timer, but I saw this isn't available on ESPHome yet), although the data comes through on the logs as minutes I think, once the timer has been set using the IR remote.

What's not working:
Setting ecomode (sleep?), setting temperature (my HA is in Fahrenheit, but AC expects C, maybe that's why!? - I noticed there's a setting on the tuya component to deal with temperature multipliers, what can I do there?), there was a setting on the original app to turn beeps on/off, how would I create a switch to control that datapoint which appears to be available as a boolean? The mode shows idle for fan, dehumidify. Datapoint 2 changes number when the mode is changed on the unit, a value of 0 means cooling mode. I would like to have the other modes display too, but don't know how to change this, maybe it's a change in the Tuya component? Or can it be done on the yaml, if so, how? I believe active_state_cooling_value of 1 should be dehumidify/dry and 2 should be fan. I would prefer to set fan speed and mode (cool/dry/fan) on the climate component rather than having two separate components, is that possible? I could compromise on having a fan entity and a climate entity BUT if I set the fan stuff, it should set the mode on the climate entity to FAN automatically, however the oscillation works for fan AND cool, so having the oscillation switch on the fan entity is not ideal.

user897943, 2022-06-08:
I have mostly good news. It's mostly working!

Here's the yaml and what I'm getting:
```
esphome:
  name: airconditioner-fb1
  platform: ESP32
  board: widora-air

wifi:
  ssid: "xxx"
  password: "xxx"
  manual_ip:
    static_ip: 10.0.0.151
    gateway: 10.0.0.1
    subnet: 255.255.255.0
    dns1: 10.0.0.1
    dns2: 10.0.0.120  

# Enable Home Assistant API
api:
  password: !secret passapi
  
ota:
  password: !secret passota

# Enable logging
logger:
  level: DEBUG
  baud_rate: 0

uart:
  - id: uart_1
    tx_pin: 1
    rx_pin: 3
    baud_rate: 9600

  - id: uart_2
    tx_pin: 17
    rx_pin: 16
    baud_rate: 9600

tuya:
  uart_id: uart_2

climate:
  - platform: tuya
    name: "Air conditioner - Front room"
    switch_datapoint: 1
    supports_heat: false
    supports_cool: true
    target_temperature_datapoint: 3
    current_temperature_datapoint: 4
    eco_datapoint: 7
    active_state_datapoint: 2
    active_state_cooling_value: 0

fan:
  - platform: "tuya"
    name: "Air conditioner (Fan) - Front room"
    speed_datapoint: 5
    switch_datapoint: 1
    oscillation_datapoint: 6
```
These are working: turning on/off from the climate component. Turning on/off from the fan component. Setting fan speed. Setting oscillation. There's a datapoint 9 (I believe) for timer, but I saw this isn't available on ESPHome yet), although the data comes through on the logs as minutes I think, once the timer has been set using the IR remote.

What's not working:
Setting ecomode (sleep?), setting temperature (my HA is in Fahrenheit, but AC expects C, maybe that's why!? - I noticed there's a setting on the tuya component to deal with temperature multipliers, what can I do there?), there was a setting on the original app to turn beeps on/off, how would I create a switch to control that datapoint which appears to be available as a boolean? The mode shows idle for fan, dehumidify. Datapoint 2 changes number when the mode is changed on the unit, a value of 0 means cooling mode. I would like to have the other modes display too, but don't know how to change this, maybe it's a change in the Tuya component? Or can it be done on the yaml, if so, how? I believe active_state_cooling_value of 1 should be dehumidify/dry and 2 should be fan. I would prefer to set fan speed and mode (cool/dry/fan) on the climate component rather than having two separate components, is that possible? I could compromise on having a fan entity and a climate entity BUT if I set the fan stuff, it should set the mode on the climate entity to FAN automatically, however the oscillation works for fan AND cool, so having the oscillation switch on the fan entity is not ideal.

user897943, 2022-06-08:
Latest logs:
```
INFO Reading configuration /config/airconditioner-fb1.yaml...
INFO Starting log output from 10.0.0.151 using esphome API
INFO Successfully connected to 10.0.0.151
[20:56:54][I][app:102]: ESPHome version 2022.6.0-dev compiled on Jun  8 2022, 20:46:12
[20:56:54][C][wifi:491]: WiFi:
[20:56:54][C][wifi:353]:   Local MAC: A4:E5:7C:52:89:A0
[20:56:54][C][wifi:354]:   SSID: [redacted]
[20:56:54][C][wifi:355]:   IP Address: 10.0.0.151
[20:56:54][C][wifi:357]:   BSSID: [redacted]
[20:56:54][C][wifi:358]:   Hostname: 'airconditioner-fb1'
[20:56:54][C][wifi:360]:   Signal strength: -52 dB ▂▄▆█
[20:56:54][C][wifi:364]:   Channel: 11
[20:56:54][C][wifi:365]:   Subnet: 255.255.255.0
[20:56:54][C][wifi:366]:   Gateway: 10.0.0.1
[20:56:54][C][wifi:367]:   DNS1: 10.0.0.1
[20:56:54][C][logger:274]: Logger:
[20:56:54][C][logger:275]:   Level: DEBUG
[20:56:54][C][logger:276]:   Log Baud Rate: 0
[20:56:54][C][logger:277]:   Hardware UART: UART0
[20:56:54][C][uart.arduino_esp32:107]: UART Bus:
[20:56:54][C][uart.arduino_esp32:108]:   TX Pin: GPIO1
[20:56:54][C][uart.arduino_esp32:109]:   RX Pin: GPIO3
[20:56:54][C][uart.arduino_esp32:111]:   RX Buffer Size: 256
[20:56:54][C][uart.arduino_esp32:113]:   Baud Rate: 9600 baud
[20:56:54][C][uart.arduino_esp32:114]:   Data Bits: 8
[20:56:54][C][uart.arduino_esp32:115]:   Parity: NONE
[20:56:54][C][uart.arduino_esp32:116]:   Stop bits: 1
[20:56:54][C][uart.arduino_esp32:107]: UART Bus:
[20:56:54][C][uart.arduino_esp32:108]:   TX Pin: GPIO17
[20:56:54][C][uart.arduino_esp32:109]:   RX Pin: GPIO16
[20:56:54][C][uart.arduino_esp32:111]:   RX Buffer Size: 256
[20:56:54][C][uart.arduino_esp32:113]:   Baud Rate: 9600 baud
[20:56:54][C][uart.arduino_esp32:114]:   Data Bits: 8
[20:56:54][C][uart.arduino_esp32:115]:   Parity: NONE
[20:56:54][C][uart.arduino_esp32:116]:   Stop bits: 1
[20:56:54][C][tuya.climate:140]: Tuya Climate 'Air conditioner - Front room'
[20:56:54][C][tuya.climate:142]:   Switch has datapoint ID 1
[20:56:54][C][tuya.climate:144]:   Active state has datapoint ID 2
[20:56:54][C][tuya.climate:146]:   Target Temperature has datapoint ID 3
[20:56:54][C][tuya.climate:148]:   Current Temperature has datapoint ID 4
[20:56:54][C][tuya.climate:152]:   Eco has datapoint ID 9
[20:56:54][C][tuya.fan:051]: Tuya Fan 'Air conditioner (Fan) - Front room'
[20:56:54][C][tuya.fan:147]:   Speed: YES
[20:56:54][C][tuya.fan:148]:   Speed count: 3
[20:56:54][C][tuya.fan:151]:   Oscillation: YES
[20:56:54][C][tuya.fan:053]:   Speed has datapoint ID 5
[20:56:54][C][tuya.fan:057]:   Oscillation has datapoint ID 6
[20:56:54][C][mdns:084]: mDNS:
[20:56:54][C][mdns:085]:   Hostname: airconditioner-fb1
[20:56:54][C][ota:085]: Over-The-Air Updates:
[20:56:54][C][ota:086]:   Address: 10.0.0.151:3232
[20:56:54][C][ota:089]:   Using Password.
[20:56:54][C][api:138]: API Server:
[20:56:54][C][api:139]:   Address: 10.0.0.151:6053
[20:56:54][C][api:143]:   Using noise encryption: NO
[20:56:54][C][tuya:033]: Tuya:
[20:56:54][C][tuya:048]:   Datapoint 1: switch (value: ON)
[20:56:54][C][tuya:054]:   Datapoint 2: enum (value: 0)
[20:56:54][C][tuya:050]:   Datapoint 3: int value (value: 18)
[20:56:54][C][tuya:050]:   Datapoint 4: int value (value: 17)
[20:56:54][C][tuya:054]:   Datapoint 5: enum (value: 0)
[20:56:54][C][tuya:048]:   Datapoint 6: switch (value: ON)
[20:56:54][C][tuya:050]:   Datapoint 7: int value (value: 0)
[20:56:54][C][tuya:056]:   Datapoint 8: bitmask (value: 0)
[20:56:54][C][tuya:048]:   Datapoint 9: switch (value: OFF)
[20:56:54][C][tuya:048]:   Datapoint 10: switch (value: ON)
[20:56:54][C][tuya:056]:   Datapoint 50: bitmask (value: 0)
[20:56:54][C][tuya:068]:   Product: ''
[... 94 lines cut ...]
[21:01:22][D][climate:399]:   Preset: NONE
[21:01:22][D][climate:408]:   Current Temperature: 17.00°C
[21:01:22][D][climate:414]:   Target Temperature: 17.00°C
[21:01:23][D][tuya:288]: Datapoint 3 update to 18
[21:01:23][D][climate:385]: 'Air conditioner - Front room' - Sending state:
[21:01:23][D][climate:388]:   Mode: COOL
[21:01:23][D][climate:390]:   Action: COOLING
[21:01:23][D][climate:399]:   Preset: NONE
[21:01:23][D][climate:408]:   Current Temperature: 17.00°C
[21:01:23][D][climate:414]:   Target Temperature: 18.00°C
[21:01:48][D][tuya:288]: Datapoint 3 update to 17
[21:01:48][D][climate:385]: 'Air conditioner - Front room' - Sending state:
[21:01:48][D][climate:388]:   Mode: COOL
[21:01:48][D][climate:390]:   Action: COOLING
[21:01:48][D][climate:399]:   Preset: NONE
[21:01:48][D][climate:408]:   Current Temperature: 17.00°C
[21:01:48][D][climate:414]:   Target Temperature: 17.00°C
[21:01:49][D][tuya:288]: Datapoint 3 update to 18
[21:01:49][D][climate:385]: 'Air conditioner - Front room' - Sending state:
[21:01:49][D][climate:388]:   Mode: COOL
[21:01:49][D][climate:390]:   Action: COOLING
[21:01:49][D][climate:399]:   Preset: NONE
[21:01:49][D][climate:408]:   Current Temperature: 17.00°C
[21:01:49][D][climate:414]:   Target Temperature: 18.00°C
```

user897943, 2022-06-09:
@jesserockz I noticed you said you were gonna implement a function (https://github.com/esphome/esphome/pull/3469), are you able to refactor the climate component to align with the climate device? Because right now, it's ridiculous - I have a select to change mode, a fan to change speed and swing, and a climate to adjust temp and turn on/off... this is crazy! It should all be under the climate component. Looks like you may have enough coding experience to pull this one off, are you able to try? Thanks.

jesserockz, 2022-06-14:
Wow, if I saw this issue when you first created it, I would have instantly suggested trying a tuya climate platform due to the `55 aa` header in the serial =)

As for the speed/swing, it needs somebody with a device to implement and test it. I do not have such a device and therefore could not test my own work, which ends up meaning it uses a lot more of my time to write code and wait for feedback to make changes.

jesserockz, 2022-06-14:
the `thermostat` climate platform is basically a `template` platform...

user897943, 2022-06-14:
I don't know how to do it, and my brain is fried after all this. If nobody would like to "fix" as a climate device with all relevant functions as it should be, I'll just use what I created but I really dislike how horrible it looks on the frontend... Anyway if someone wants to tackle this last thing I will test and they'll certainly have my full gratitude.

sid-the-squid, 2023-01-04:
Hi all, 
Thanks to this thread I purchased a Princess 9000 AC unit, in June 2022, just before it got crazy hot here in the UK.. However my first child was also born that month leaving me no time to hack around with the unit.. until now.

You guys did a great job of figuring out how this works, and the solution figured out by OP was certainly good enough on the functionality side.

'jesserockz' comment about thermostat being a template platform made me think that perhaps there is another way this can be approached without writing a custom cpp component.

So I've come up with this potential solution, which @user897943  may find useful as it presents in HA as desired, and seems to function as expected.

I've shoved the yaml in my github repo, link here [github,](https://github.com/sid-the-squid/princess_ac/blob/main/princess_mobile_ac.yaml) please note that this yaml is written for an ESP8266 as I could not make the factory ESP32 function, so take the bits out that are relevant to you.

hope this as useful to you as this thread was to me :)