esphome/feature-requests #1737: Add support for Haxxair AC remote model DG11J1-72 in IR Remote Climate

workingmanrob, 2022-05-19:
**Describe the problem you have/What new integration you would like**

Control Haxxair AC with IR Remote Climate - remote model DG11J1-72

**Please describe your use case for this integration and alternatives you've tried:**

Haxxair split head AC unit seems to use a similar remote to the ones listed under whirlpool.  Remote model DG11J1-72

This unit is at another location so I have not yet tested any of the existing whirlpool remote options or Arduino-HeatpumpIR.

**Additional context**

You guys rock!

graescale, 2022-05-19:
Yes please! I second that!

workingmanrob, 2022-08-06:
Finally managed to get to the same location as the AC unit with an esp8266 in whirlpool mode but unfortunately neither remote type worked out of the box.

DG11J1-3A: Temperature range is from 18 to 32 (default)
DG11J1-91: Temperature range is from 16 to 30

Again the Haxxair remote is labelled DG11J1-72 so I was hoping for common codes.

With nothing working I tried to capture the power button codes but the unit still didn't respond `¯\_(ツ)_/¯`

```
  turn_on_action:
    - remote_transmitter.transmit_lg:
        data: 0xC16024CE
        nbits: 32
    - remote_transmitter.transmit_nec:
        address: 0x0683
        command: 0x7324
    - remote_transmitter.transmit_jvc:
        data: 0xC160
  turn_off_action:
    - remote_transmitter.transmit_lg:
        data: 0xC16024CE
        nbits: 32
    - remote_transmitter.transmit_nec:
        address: 0x0683
        command: 0x7324
    - remote_transmitter.transmit_jvc:
        data: 0xC160
```
If a full dump of the remote codes would help I can collect that next time and attach it here.