esphome/feature-requests #1741: Request for friendly name for ESPHOME devices to use HA feature to reduce entity names

Bascht74, 2022-05-23:
**Describe the problem you have/What new integration you would like**
I can assign the esphome device a name with: 

```
substitutions:
  upper_devicename: "Lampe Aussen Hausbeleuchtung"
  devicename: "lampe-aussen-hausbeleuchtung" #(upper case and space not allowed)
esphome:
  name: $devicename
```

but I cannot assign a friendly name for Home Assistant: 
`friendly_name: ${upper_devicename} #--> doesn't work`

If I give e.g. a light entity a name: 
```
light:
  - platform: binary
[...]
    name: "${upper_devicename} Balkon"
```

It doesn't show the right way in Home Assistant: 
[image]

If I rename the device name into the "friendly one" it works: 
[image]

So it would be nice to have a "friendly" device name with uppercase and spaces via the yaml code. That way the look in HA would be much cleaner

### Anything in the logs that might be useful for us?

```txt
nothing
```

**Additional context**
It would be an alternative, if the `esphome name` accepts upper case and spaces and translates to lowe case and "-" were needed.

If you would be able to set an uppercase device name via the yaml you can make use of the truncating in Home Assistant (would be much clearer)
**Please describe your use case for this integration and alternatives you've tried:**

Example YAML snippet
```yaml
substitutions:
  upper_devicename: "Lampe Aussen Hausbeleuchtung"
  devicename: "lampe-aussen-hausbeleuchtung"

esphome:
  name: $devicename
#  friendly_name: ${upper_devicename} --> missing?

light:
  - platform: binary
    id: light1
    output: relay1
    name: "${upper_devicename} Balkon"
```

nagyrobi, 2022-05-23:
https://github.com/esphome/feature-requests/issues/1539

nagyrobi, 2022-07-01:
https://github.com/esphome/feature-requests/issues/1741

closed 2022-07-01 by nagyrobi

robloh, 2023-02-07:
It looks like this is coming in the next release: https://github.com/esphome/esphome/pull/4296

nagyrobi, 2023-02-07:
hooraaayy

vincekovacs, 2023-02-25:
As I see, this pull request only allow to add friendly name to device but not for individual entities (switches, lights, inputs, etc).

It is possible to have an entity identifier used in HA regardless of the name?

Like this:

```yaml
light:
  - platform: binary
[...]
    name: "${upper_devicename} Balkon" // <-- display name in HA
    entity_id: "${devicename}_balkon" // <-- used in HA
```

bonzai86, 2023-02-28:
+1

I'd love to rename a entity without breaking all dashboards/automations or manually rename within HA.

EmhyrVarEmreis, 2023-11-16:
+1
I see that sth was merged. Any idea when it will become available?

functionpointer, 2023-12-05:
Friendly name is also used if you set
```
name: None
```
This marks the entity as the primary for the device. HA will show the friendly name as the name.
ESPHome also uses the friendly name as entity id, and HA doesn't add anything.

This pattern isn't continued when a real name is set. We still can't re-use a `name` for different devices, the unique_id will clash.

jougs, 2025-01-06:
See also #2119.