esphome/feature-requests #1743: Add the support of IR Remote Climate for De'Longhi Penguino PAC EX100 (IR mapping provided)

idefxH, 2022-05-24:
**Describe the problem you have/What new integration you would like**

Hello, I'd like to use my AC with IR Remote Climate, but it is not supported yet.
It is a De'Longhi Penguino PAC EX100, and here is the mapping with IR messages (all messages are nec format)

```
POWER (on/off):                 address=0xFF1F, command=0x7C83
Switch between modes :          address=0xFF1F, command=0x3EC1 #Climatisation, dehumidification, Fan only
Switch between fan levels:      address=0xFF1F, command=0x3FC0 #Climatisation mode: 4 modes, low, medium, high, auto. Fan mode: only low, medium and high
+ :                             address=0xFF1F, command=0x7B84 #Climatisation mode: From 18°c to 32°c, incr. 1. Timer setting: from 0.5h to 24h, incr .5h until 9.5, then incr 1
- :                             address=0xFF1F, command=0x7F80 #Climatisation mode: From 18°c to 32°c, incr. 1. Timer setting: from 0.5h to 24h, incr .5h until 9.5, then incr 1
Timer setting:                  address=0xFF1F, command=0x7E81 #Enter timer setting mode when activated. Validated if nothing is pressed for few seconds
Switch between air-flow modes:  address=0xFF1F, command=0x7A85 #2 modes. Static or moving
Auto mode:                      address=0xFF1F, command=0x7D82 #Automatic mode, no setting available (except Timer)
Silence mode switch:            address=0xFF1F, command=0x7986 #Override current fan level. If fan mode is click when in silent mode, fan mode reverts to previous setting
```
**Please describe your use case for this integration and alternatives you've tried:**

Doesn't work with existing delonghi IR Remote Climate integration (tested, not same commands nor address)

**Additional context**

N/A

Px7-941, 2024-07-28:
Is there a way to use these IR commands in esphome?
How would I have to integrate it before we have "native" esphome support?

idefxH, 2024-07-30:
Yes, I did something quite nasty, too nasty to contribute back, but it involves many things (
- an esphome device with IR
```
esphome:
  name: climate-ir

esp32:
  board: pico32
  framework:
    type: arduino

# Enable logging
logger:

esp32_ble_tracker:

globals:
  - id: ongoing_transmission
    type: bool
    restore_value: no
    initial_value: 'false'

      
remote_transmitter:
  id: trsmt
  pin: GPIO26
  carrier_duty_percent: 50%

bluetooth_proxy:

remote_receiver:
  id: rcvr
  pin: 
    number: GPIO32
    inverted: true
    mode:
      input: true
      pullup: true
  tolerance: 55%
  on_nec: 
    - then:
      - logger.log: 
          format: "Recieved nex: %X"
          args: 'x.command'
      - if:
          condition:
            lambda: 'return x.command == 0x9999;'
          then:
            - logger.log: "set ongoing_transmission to true"
            - globals.set:
                id: ongoing_transmission
                value: 'true'
          else:
            - if:
                condition:
                  lambda: 'return x.command == 0xFFFF;'
                then:
                  - logger.log: "set ongoing_transmission to false"
                  - globals.set:
                      id: ongoing_transmission
                      value: 'false'
                else:
                  - logger.log: 
                      format: "codix recieved, but wrong code: %X"
                      args: 'x.command'
                  - if:
                      condition: 
                        lambda: 'return id(ongoing_transmission);'
                      then: 
                        - logger.log: "ongoing sync, sorry"
[... 431 lines cut ...]
        previous_task.cancel()
    except NameError as e: #Cool, we can start
         pass

    task.unique("clim_sync" + from_value, kill_me = False)
   

    task.sleep(1)
    update_clim_mode(from_value)

    ghost_clim_mode = get_clim_mode()

    update_silence(from_value, ghost_clim_mode)

    update_fan(from_value, ghost_clim_mode)

    update_temp(from_value, ghost_clim_mode)

    update_swing(from_value, ghost_clim_mode)

```

And other fun stuff in your home assistant configuration that would take quite some time for me to isolate and share, as I did it > 1 year ago.

I should probably create a github project out of it, but I don't have much time right now. If you are interested, we can work on it together?