esphome/feature-requests #1747: Bedjet - add automation for lambdas @jhansche

user897943, 2022-05-28:
**Describe the problem you have/What new integration you would like**

@jhansche requested to open a feature request to add automation for lambdas to expose other data values from the bedjet unit to HA as sensors for example.

See discussion from this point down: https://github.com/esphome/issues/issues/3286#issuecomment-1139915913
**Please describe your use case for this integration and alternatives you've tried:**

Bedjet - see above
**Additional context**

user897943, 2022-06-13:
@jhansche sorry for the unrelated question. I am using the ESP board to exclusively run this integration. I saw this: https://medium.com/@qz_li/smart-bed-7de9ad55276e and would like to run both the BedJet stuff AND this, since the pins needed to achieve the project are available on the ESP and unaffected by the BedJet stuff... So could I put the code relevant to the sensing into the yaml and plug the wires, thus having both bed sensing and bedjet control? I wanna check first with you in case it will cause any conflicts.

jhansche, 2022-06-14:
@user897943 The `esp32_touch` component appears to be pretty lightweight, so I do not expect this will cause any problems.

I am not as confident adding MQTT to the mix, however - but the guide seems to imply that MQTT is an optional replacement for the HA API (`api:` component in yaml). So it sounds like if you use MQTT you would probably also disable `api:`, and then maybe that's ok.

I have not used MQTT alongside a BLE component, but the BLE specs do specifically warn about adding a lot of components and that it might introduce instability. I haven't done it myself, but there are reports that it can work. All I can say is that I personally cannot vouch for BLE + MQTT together.

user897943, 2022-06-14:
> @user897943 The `esp32_touch` component appears to be pretty lightweight, so I do not expect this will cause any problems.
> 
> I am not as confident adding MQTT to the mix, however - but the guide seems to imply that MQTT is an optional replacement for the HA API (`api:` component in yaml). So it sounds like if you use MQTT you would probably also disable `api:`, and then maybe that's ok.
> 
> I have not used MQTT alongside a BLE component, but the BLE component docs do specifically warn about adding a lot of components and that it might introduce instability. I haven't done it myself, but there are reports that it can work. All I can say is that I personally cannot vouch for BLE + MQTT together.

Thanks, I'm not using MQTT, I'm going to use the API. I bet you're going to try this too, won't you?