esphome/feature-requests #1749: Add update entity for ESPHome device(s)

BertVereecke, 2022-05-30:
**Describe the problem you have/What new integration you would like**
The `update.esphome_update` entity indicates whether the ESPHome add-on is up-to-date, and provides a quick way to see the current and latest version of the add-on.
Would it be possible to add update entities for the ESPHome devices too?

**Please describe your use case for this integration and alternatives you've tried:**
Update entities provide a quick way to see which components are running on the latest version, and which ones are outdated.
The device versions can be seen in the ESPHome Web UI, and can be retrieved via `{{ device_attr(device_id('devicename'), 'sw_version') }}`, but having an update entity readily available for all ESPHome devices would be much easier.

**Additional context**
n/a

crixionz, 2022-08-17:
Hi @BertVereecke, I was browsing through feature requests and saw yours a few hours ago. I cannot help in making an update entity, but I saw your template and it made me dig today's rabbit hole and I ended up with this template. 
It prints all your ESPHome devices' versions and datetimes into a table, sorted by date of installation.
Maybe it helps bridge the time until your FR is implemented. :)
```jinja2
{% set x = namespace(devices=[]) %}
{% for device in integration_entities('esphome')|map('device_id')|unique|list -%}
  {% set x.devices = x.devices + [{
  'name':(device_attr(device, 'name')|title)[:14],
  'timestamp':strptime(device_attr(device, 'sw_version')[-23:],'(%b %d %Y, %H:%M:%S)')|as_timestamp|int,
  'version':device_attr(device, 'sw_version')[:-24], }] %}
{%- endfor -%}

{% for i in x.devices|sort(attribute="timestamp",reverse=false) -%}
  {{ ' '*(14-i.name|length) ~ i.name ~ ':\t   ' ~ i.version ~ '\t' ~ i.timestamp|timestamp_custom('%Y.%m.%d  %H:%M') ~ '\n' }}
{%- endfor %}
```

felipecrs, 2022-10-27:
Yeah, this would definitely be awesome!

Unfocused, 2024-03-20:
This was added in Home Assistant 2023.2.0 and ESPHome 2023.2.0.

See the announcement here: https://www.home-assistant.io/blog/2023/02/01/release-20232/#now-even-tighter-integrated-with-esphome

kpfleming, 2024-06-26:
@nagyrobi This one appears to be completed too :-)

closed 2024-06-26 by nagyrobi