esphome/feature-requests #1751: Add support for audio volume controllers like DS4420

nagyrobi, 2022-05-31:
**Describe the problem you have/What new integration you would like**
With the new media player functionality added to ESPHome, audio volume control is another thing to tkink about.

DS4420 is a good solution as it allows control over I2C, and it allows up to eight devices on a common bus, which is important for implementations using [multiple auxiliary inputs](https://github.com/esphome/feature-requests/issues/1750) where each input needs its own gain level, and a general master output level could be handy too.

**Please describe your use case for this integration and alternatives you've tried:**
Manipulating digital signals mathematically to adjust audio levels is a complex thing if you don't want severe sound quality degratation and is likely out of scope for ESP32 especially due to limited resources. Also for auxiliary sources which could be switched in plain analog, a programmable gain amp is the easiest and cleanest way to go.

**Additional context**
Related to https://github.com/esphome/esphome/pull/3487

nagyrobi, 2022-06-17:
Use case: https://forum.raspiaudio.com/t/suggestion-for-espmuse-multiple-analog-inputs/401

dtill, 2024-07-25:
id love to see this! generally speaking esphome lacks badly of a DEVICE_CLASS_AUDIO_VOLUME DEVICE_CLASS_AUDIO_EQUALIZER. 

just audio control without caring about media players.  then people can start building implementing components for all the great audio gear out there :)) that be awsome! the audiophile people havne't arrived to the esphome ecosystem ..

 this should be available up to the frontend. Sound is a basic output and ambient feature as light is and everybody uses it. these components  should be adjustable like color light components  (but with optionally eqalizable sound instead of colors) 
media player entities could then use the se components for their volume control as well. 
also muting sound globally in a home automation setup can make very much sense. so there should be a general event mute all volume controls should listen to. 

its can be kind of a subset of the media_payler component now., so the volume_control capabilities can be sources out of the media player component.

sry for brainstorming here.. hope someone comes across here and finds himself building nice volume knobs for the web server frontend and HA dashboard as well. that be a dream!