esphome/feature-requests #1753: Full Daikin Controller Replacement

urbozz, 2022-06-02:
**Describe the problem you have/What new integration you would like**

Daikin's newest controllers (such as the BRP069C41) require a cloud account and cannot be controlled locally. Most split units the controller connects to either a S21 port/protocol (older and as the BRP069B41 uses) and is well documented, or newer connector/protocol as the BRP069C41 uses.

Both of these types of connector have been reversed engineered and implemented so that an ESP32 can fully replace the Daikin Controllers, [ESP32-Daikin](https://github.com/revk/ESP32-Daikin) with some excellent blog posts of the reverse engineering process at [RevK®'s ramblings](https://www.revk.uk/)

Would it be possible to use these findings and implement this in ESPHome, so ESPHome and an ESP32 can be used to fully replace Daikin's online controllers?

**Please describe your use case for this integration and alternatives you've tried:**

**Additional context**

grob6000, 2022-08-13:
+1 for this, would love to convert my two esphome converted sensibos into internal controllers that interface directly with the unit!

Quarco, 2022-09-06:
This would be awesome! +1

jonchaka, 2022-12-03:
+1 for this

Quarco, 2022-12-03:
+1 indeed!

BradleyFord, 2023-01-18:
+1 from over here as well

estqwerty, 2023-01-20:
waiting for this one. anyone tried to build this ?

asund, 2023-01-21:
I just put down a deposit on a ductless Daikin system and will check in and look into doing this if no one else has in the future. Got a lot of more important home improvement tasks to do first though.

joshbenner, 2023-05-19:
I've made a custom component available on GitHub that does this with ESPHome: https://github.com/joshbenner/esphome-daikin-s21