esphome/feature-requests #1754: Option to log to file from the Esphome UI

RoboMagus, 2022-06-03:
**Describe the problem you have/What new integration you would like**

For rare (once in a long time) issues it would be a really nice feature to be able to have the log of EspHome devices redirected to a file. Much like you can now already wirelessly look at the live logs, It would be a great feature to have the option to 'store the log of the device to a file' added to the choices in the `How to get the logs for your device?` menu. 
I would then also envision another button next to the `secrets` button in the top of the UI that allows you to look at (or download) the currently stored logfiles on the EspHome server.

**Please describe your use case for this integration and alternatives you've tried:**

The use for this would of course be capturing the logfiles of the EspHome devices from within the docker container, which should later on be viewable through the UI. This would aid in debugging hard to track problems that only occur once every so often.

I am currently doing this through CLI in the docker container by running `esphome logs device.yaml >> device.log`. But it would be so much better to have the option to do so through the web interface.

**Additional context**

Additional things to consider when adding this feature could be:
 * Start / stop buttons for logging to file
 * Storing each devices logfile in a date-time formatted filename
 * Auto cleanup of old logfiles?
 * Start logging for _x_ amount of time?

RoboMagus, 2022-07-08:
Found [this thread](https://community.home-assistant.io/t/esphome-logger-to-a-file/245946/4) on logging to file.

backcountrymountains, 2023-04-06:
I came here to give this feature request a thumbs up.
I had to ssh into HA run:
```docker exec -it addon_15ef4d2f_esphome /bin/bash``` 
to get into the esphome addon container and then
```/config/esphome# esphome bn3.yaml logs --device OTA >bn3.log``` from ```/config/esphome/``` to redirect the logs to a file.

A button to do that would be nice.

jrob, 2024-06-21:
I'd love to see something like this. I have an issue with a couple of custom components that give me problems occasionally. Could be fine for a couple of months, then happen twice in a week. Leaving a window open or a custom command running in the container isn't really a viable option, as the issues are super sporadic. I really only have a couple of requirements. 

1. Auto log to a plain old text file on homeassistant.
2. Trim after size or date.

Bonus points for a viewer that shows the text in the same color as the component output.

pixelwave, 2024-06-22:
> How can I stop the logging to file?

Ok I figured it out:

```
#SSH into esphome docker
#install ps (optional)
apt-get update && apt-get install procps

#list running esphome processes
ps aux | grep esphome

#kill running process (via PID)
kill <PID>
```