esphome/feature-requests #1756: Add support for Soil Moisture Sensor (SoMoSe) from BeFle

rasda, 2022-06-04:
**Describe the problem you have/What new integration you would like**

I bought a Soil Moisture Sensor from BeFlE which is called SoMoSe (v2.2).
https://github.com/BeFlE/SoMoSe
I got it working using the analog Pin of the Sensor and have readings in Esphome/HomeAssistant.
But i was not able to integrate the digital I²C Sensor in Esphome as i am not so skilled with programming.
I read the Documentation about Custom Sensors but again ... i am not a programmer and did not succeed.

**Please describe your use case for this integration and alternatives you've tried:**

Get Soil Moisture Readings for my Chiliplants via the digital Sensor (I²C) from BeFlE SoMoSe Sensor.
Analog is working but not very accurate.

**Additional context**

BurneyFM, 2023-05-06:
I made the same experience - wolle love to see this happen!

Parateam, 2024-04-19:
Hello, i have 6 somose Sensors in a project connected with i2c.

Its working with the esphome-sample in the github repository from somose.

BUT, when 2 or more of the 6 sensors are not connected, the ESP32 doesen't boot any more. 

Can someone look at that? My knowledge is not high enough.