esphome/feature-requests #1759: We have "on_value" and "on_raw_value" but *no* "on_raw_value_range"

Bascht74, 2022-06-06:
**Describe the problem you have/What new integration you would like**
I would like to have a temperature sensor, that is updated only every 30 or 60 seconds in home assistant. On the other side, I want to turn a switch off immediately if the temperature is above a value. 

so I got this: 

```
  - platform: ntc
    sensor: temp_resistance_reading
    name: "${devicename} Temperature"
    unit_of_measurement: "°C"
    accuracy_decimals: 1
    icon: "mdi:thermometer"
    calibration:
      b_constant: 3350
      reference_resistance: 10kOhm
      reference_temperature: 25°C
    on_value_range:
      - above: ${max_temp}
        then:
          - switch.turn_off: "relay"
    filters:
      - throttle: 60s
```

But in this case the filters (=throttle) are processed so that on_value_range is upmost 60s late.
There ist `on_raw_value` but no `on_raw_value_range`.

**Please describe your use case for this integration and alternatives you've tried:**
I would like to have a `on_raw_value_range` to check the temperature very frequently but to send the sensor to HA only from time to time.

**Additional context**
I would like to use that for my Shelly 1PM / 2.5 / plus 1PM and more...

nagyrobi, 2022-06-09:
You could use the [copy](https://esphome.io/components/copy.html) platform just for that. You move the `throttle` filter and the UI properties to the copy and publish that instead. Like:

```
  - platform: ntc
    sensor: temp_resistance_reading
    id: temp_sens
    internal: true
    accuracy_decimals: 1
    calibration:
      b_constant: 3350
      reference_resistance: 10kOhm
      reference_temperature: 25°C
    on_value_range:
      - above: ${max_temp}
        then:
          - switch.turn_off: "relay"

  - platform: copy
    source_id: temp_sens
    name: "${devicename} Temperature"
    unit_of_measurement: "°C"
    accuracy_decimals: 1
    icon: "mdi:thermometer"
    filters:
      - throttle: 60s
```

bkaufx, 2022-06-09:
You could also just use on_raw_value, and check if the temp is above max within that automation.

Bascht74, 2022-06-10:
Thx for your infos. That helps!

closed 2022-06-10 by Bascht74